micropython/micropython-lib #18: Conventions for MicroPython -> CPython compatibility modules

pfalcon, 2015-01-01:
The implementation of this probably belongs to micropython-lib, but as we have more people here, I post RFC here.

So, as we go forward, uPy offers more and more features which are not compatible with CPython. But we surely want to remain good members of Python community, and provide a compatibility layer to run MicroPython-optimized programs on other Python implementations. We had beginnings of that for a long time - https://github.com/micropython/micropython/blob/master/examples/micropython.py , and this RFC is to set conventions and guidelines how to go forward with this.

My proposals:

1. Host such compatibility modules in micropython-lib repo.
2. To distinguish them from native uPy modules, have their dir names prefixed with "cpython-".
3. For PyPi, stick with existing convention of "micropython-" prefix, which them will be followed by "cpython-" infix.

As an example, CPython compatibility shim for MicroPython's "micropython" module will be possible to install from PyPi as "micropython-cpython-micropython".

pfalcon, 2015-01-03:
First compatibility module added: https://github.com/micropython/micropython-lib/tree/master/cpython-uasyncio

dlech, 2017-09-19:
What would it take to get CPython compatible versions of the `micropython`, `framebuf` and `uctypes` modules?

jonnor, 2024-08-25:
The current policy is that CPython-type modules should have the same, and be as compatible as possible with the matching CPython module. With MicroPython 1.21 all the `usomething` modules where renamed to just `something`. https://github.com/orgs/micropython/discussions/12602

closed 2024-08-25 by robert-hh