micropython/micropython-lib #22: Request for inspect module to be ported

Ivoah, 2015-02-17:
Request for inspect module to be ported

dpgeorge, 2015-02-17:
This is a difficult one to do, since MicroPython has very different internals to CPython.  For example, sometimes there is no bytecode!  What particular features of inspect do you need?

Ivoah, 2015-02-17:
I want to get sympy working with micropython, and it uses the inspect module
I think these are all the functions it uses:
* getmro
* getargspec
* currentframe
* getsourcefile
* getsource
* isfunction
* isclass
* isgeneratorfunction
* ismodule
* getframeinfo
* getfile

dpgeorge, 2015-02-18:
To see the full list of dependencies for sympy, try running this script:

```python
orig_import = __builtins__.__import__
imp_depth = 0
def __my_import__(name, globals=None, locals=None, fromlist=(), level=0):
    global imp_depth
    print('[IMPORT]', ' ' * imp_depth, name)
    imp_depth += 1
    try:
        mod = orig_import(name, globals, locals, fromlist, level)
    finally:
        imp_depth -= 1
    return mod

__builtins__.__import__ = __my_import__

# import here the modules you want to see the dependency graph for
import numpy
```

Ivoah, 2015-02-23:
Here is a pastebin of the sympy imports: http://pastebin.com/fav1hr8K

prologic, 2015-04-26:
Apparently I need this too for [circuits](https://github.com/circuits/circuits) on the [MicroPython + WiPy](https://github.com/circuits/circuits/milestones/MicroPython%20+%20WiPy)

Specific things from ``inspect`` that we probably need are:

```#!bash
prologic@daisy
Mon Apr 27 08:38:31 
~/circuits
$ grin inspect circuits
circuits/core/components.py:
   11 : from inspect import getmembers
circuits/core/events.py:
   11 : from inspect import ismethod
circuits/core/handlers.py:
    9 : from inspect import getargspec
circuits/core/loader.py:
   12 : from inspect import getmembers, getmodule, isclass
circuits/core/manager.py:
   13 : from inspect import isfunction
circuits/tools/__init__.py:
  128 : def inspect(x):
  129 :     """Display an inspection report of the Component or Manager x
  134 :     @return: A detailed inspection report of x
circuits/web/controllers.py:
   11 : from inspect import getargspec
circuits/web/exceptions.py:
   12 : from inspect import isclass
circuits/web/main.py:
   29 : from circuits.tools import inspect, graph
  227 :         print(inspect(manager))
```

* ``from inspect import getmembers``
* ``from inspect import ismethod``
* ``from inspect import getargspec``
* ``from inspect import getmodule``
* ``from inspect import isclass``
* ``from inspect import isfunction``
* ``from inspect import isclass``

At this point I'm not sure how much of this can be gleaned from the MicroPython runtime at all. It's worth looking in to :)

pfalcon, 2015-04-28:
@prologic: Thanks for posting about your interest. What I can reply so is mostly: ack. And it was on my plan to provide (mostly dummy) implementation of inspect. Now that more than one person asked for it, i'll try to bump the priority. 

But:

> At this point I'm not sure how much of this can be gleaned from the MicroPython runtime at all.

The best approach is not to "glean" that stuff form MicroPython runtime, but to make you framework need as little as possible that info. Otherwise, you unlikely will be able to run on embedded targets (introspection requires just too much memory); you unlikely will be able to enjoy native codegeneration speed optimization.

prologic, 2015-04-28:
Yes I see your point -- There is a cost to dynamic(isms) :) Perhaps there is a compromise between some of the more dynamic features of circuits and it's general eventing

pfalcon, 2015-04-28:
> There is a cost to dynamic(isms)

Exactly. And that's the basic idea of MicroPython - you can run Python code even on MCU, you just need to say bye to over-dynamicism (and that's not property of language, but more of particular code and application - the language offers those features, but as the help to prototype and debug apps, they are not needed to run large set of production apps).

With that in mind, please see https://github.com/micropython/micropython-lib/commit/7eb7eba659854fdd9223bbd3398dc3603c856e80 . No surprises - most functions are dummy. I don't know how people are using that module (I for one never use it in my code) - my assumption is for example to give additional info in case of error. Then, the module even as it is now can help you further progress with your porting (because there're certainly more things to take care about besides inspect).

prologic, 2015-04-29:
Related: micropython/micropython#405

pfalcon, 2015-05-08:
@Ivoah, @prologic : Please add your progress/notes to this topic on forum: http://forum.micropython.org/viewtopic.php?f=5&t=104 (you can read it for inspiration too).

Closing otherwise.

closed 2015-05-08 by pfalcon