micropython/micropython-lib #25: libpcre.so.3 is not on all systems

dpgeorge, 2015-06-03:
On my Arch Linux machine `import re` fails because libpcre.so.3 does not exist.  Instead it's libpcre.so.1 (also aliased as libpcre.so).

So should we just change it to load libpcre.so, or try different versions in succession?

dhylands, 2015-06-03:
Normally, you would just load libpcre.so and it would be symlinked to the appropriate version.

I've also seen stuff where x.so -> x.so.7 -> x.so.7.4 -> x.so.7.4.3

And most code would be symlinked to x.so and would only include the version if there was a very good reason to.

pfalcon, 2015-06-03:
> Normally, you would just load libpcre.so and it would be symlinked to the appropriate version.

No, that's not how it works. "3" in libpcre.so.3 is a "major version". It's deemed that if major version changes, the lib grew "completely incompatible" with older (or future) lib version. Apps usually link to such major-specified lib versions. Now pure "libpcre.so" is there for naive ld -lpcre to work (how it comes from user specifying -lpcre but "libpcre.so.3" being recorded in the final executable is under veil of darkness, consider that linker magic). In Debian and derivatives, "libpcre.so" is part of *-dev packages. So yes, the easiest way is to load "libpcre.so", except you can't rely it's there.

pfalcon, 2015-06-03:
So, we have a method to try few lib names in turn already (for libc), can use it here too.

@dpgeorge : What actual pcre version your system has? Here it's:

libpcre3:amd64                                              1:8.31-2ubuntu2

pfalcon, 2015-06-03:
> Apps usually link to such major-specified lib versions. 

e.g.:

~~~~
$ ldd `which mc`
[]
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fc325fd7000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007fc325dd3000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fc325acd000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007fc3258b3000)
	libgcrypt.so.11 => /lib/x86_64-linux-gnu/libgcrypt.so.11 (0x00007fc325633000)
	libpcre.so.3 => /lib/x86_64-linux-gnu/libpcre.so.3 (0x00007fc3253f5000)
~~~~

dpgeorge, 2015-06-03:
It's pcre 8.37-1.  The filename is /usr/lib/libpcre.so.1.2.5

I've tested re-pcre/tests_re.py and it works.

BTW, a different solution would be to use builtin ure to implement core re stuff.

pfalcon, 2015-06-03:
> It's pcre 8.37-1. The filename is /usr/lib/libpcre.so.1.2.5

So, I would imagine, distros would leave shlib versioning to upstream projects. But I may imagine that Debian munged it much enough to patch major version (I assume that's what happens).

> BTW, a different solution would be to use builtin ure to implement core re stuff.

Feel free to start re_ure. But when would it have named capture groups? Not soon. Generic forward/backward assertions? Never. So, currently there's just ure for unbloated usage and re_pcre for real-world usage (pcre is already incompatible in minor details to cpython's SRE and apps need patching here and there).

dpgeorge, 2015-06-03:
> Feel free to start re_ure.

How would you select between the two (if you had both installed)?

[Background: I'm using picoweb to write a simple server and it depends on re.  ure is good enough as it is now for my purposes.]

pfalcon, 2015-06-03:
> How would you select between the two (if you had both installed)?

By using dependency injection pattern. No "library" package should depend on a particular re implementation, it should be select by a top-level user app. I believe that's how it is done already for picoweb/notes-pico, as a proof of concept.

dpgeorge, 2015-06-03:
> I believe that's how it is done already for picoweb/notes-pico, as a proof of concept.

I don't see that pattern anywhere in your code...  picoweb imports re in `__init__.py` and utils.py.

pfalcon, 2015-06-03:
Hello,

> > I believe that's how it is done already for picoweb/notes-pico, as
> > a proof of concept.
> 
> I don't see that pattern anywhere in your code...  picoweb imports re
> in `__init__.py` and utils.py.

I don't mean it literally, as a programming "dependency injection"
pattern. I mean meta-pattern, when dependencies are specified at the
top system level, not by intermediate components (and thus entire
system "injects" dependencies into its individual components).

So, both "re_pcre" and "re_ure" distribution packages (Python term)
would provide "re.py". Any intermediate components which would do
"import re" would add neither "re_pcre" nor "re_ure" to list of its
dependencies (to setup.py or requirements.txt), e.g.:
https://github.com/pfalcon/picoweb/blob/master/setup.py

Then selecting which particular implementation to use lies on the
particular top-level (as run by user) application, e.g.:
https://github.com/pfalcon/notes-pico/blob/master/requirements.txt   

> 
> ---
> https://github.com/micropython/micropython-lib/issues/25#issuecomment-108470168

dpgeorge, 2015-06-03:
> I don't mean it literally, as a programming "dependency injection" pattern. I mean meta-pattern, when dependencies are specified at the top system level

Ok, I see.  But then you need a separate installation of libraries for each app.

pfalcon, 2015-06-03:
> Ok, I see. But then you need a separate installation of libraries for each app.

Well, kinda yes, or at least, it's not possible to install both re_pcre and re_ure at the system level at the same time. But we target small self-contained apps anyway, right? ;-)

pfalcon, 2015-06-04:
Ok, any patch coming for this?

dpgeorge, 2015-06-04:
> Ok, any patch coming for this?

I can do the "try different libs" option.  We might eventually want some common code to do this rather than writing it out each time per module.

pfalcon, 2015-06-04:
See _libc.py, which has beginning of that, which apparently needs further refactoring to be reusable.

dpgeorge, 2015-06-04:
> See _libc.py, which has beginning of that, which apparently needs further refactoring to be reusable.

Obvious place for such a "load many" function is in ffi module itself.  But ffi is written in C.  We can:

1. Rename C ffi module to `_ffi` then provide Python ffi module which wraps it.
2. Add ffiext module in Python with extensions to ffi.
3. Write the "load many" feature in C.  It would be about 5 extra lines to ffimod_make_new (and would be more efficient searching than raising a Python exception on each failed file).

dpgeorge, 2015-06-04:
> Write the "load many" feature in C. It would be about 5 extra lines to ffimod_make_new

That's a bit cryptic :)  What I meant was to allow the argument to ffi.open to be a string *or* a list/tuple of strings.  In the latter case they would be tried in order until success (or raise exception if all fail).  Eg: `ffi.open(('libc.so', 'libc.so.1'))`

See also #7 for related discussion.

pfalcon, 2015-06-04:
Well, I don't know. As a pragmatic solution, we can have that function in _libc.py for now. Also, one way to deal with it is to have a lib basename, then a func try to append various prefixes/suffixes, version numbers, etc. to it (And try one by one, no need to have a list, and no need to code list support on C level).

pfalcon, 2015-06-04:
Maybe drop "_libc" and replace with "ffilib" which will be able to load any lib (with caching).

dpgeorge, 2015-06-04:
> Also, one way to deal with it is to have a lib basename, then a func try to append various prefixes/suffixes, version numbers, etc. to it

But that means we could load a random version.  Isn't the point that we want to only search for known good versions?

pfalcon, 2015-06-04:
Probably. The point is, I don't think list loading needs to be implemented on C level.

pfalcon, 2015-06-04:
> But that means we could load a random version. Isn't the point that we want to only search for known good versions?

Also "known good" is a bit moot thing. If we load libfoo.so as a fallback (and we apparently should do that), then we load not "known good", but "possibly existing". Suppose we start loading with highest major version, but what if user runs it on an ancient Debian version ("debian stable"?) with "bad" major version of 1 (which is "good" on Archlinux)?

So, it's not our task to figure out "good" version of lib, it's user's task to have it. Our task is to KISS and easy for user to hack to workaround any issues they see.

dpgeorge, 2015-06-04:
See #27 for a proposed patch.

pfalcon, 2015-06-06:
> Let's make sure we know the next steps. And that would be adding a convenience method to load libc - as I mentioned, this new module is to replace _libc.py. So, suppose we add "libc()" function for that, which saves other modules from repeating params, us patching them in multiple places, and actually, it's quite adhoc matter (guess how libc is called under Windows).

> What's next? On normal unix, socket API is part of normal API, but on windows, it's a separate DLL. So, need another convenience function in this module, which will return the same libc on unix, but adhoc lib on Windows. The idea is clear I hope. Thinking about naming convention for such functions beforehand may help then.

@dpgeorge : Any opinion on these? Should it be fflib.libc(), ffilib.libsocket(), etc?

dpgeorge, 2015-06-06:
> Should it be fflib.libc()

Yes this seems to be the most obvious thing to call it.

> ffilib.libsocket()

Yes, also fine, since the main exported function will be "socket".

jonnor, 2024-08-25:
#27  was merged, presumably fixing this. Proposing to close.

closed 2024-08-25 by robert-hh