micropython/micropython-lib #30: No chance to achieve native uasyncio->asyncio compatibility

pfalcon, 2015-06-15:
See http://bugs.python.org/issue24449

@dpgeorge : FYI

pfalcon, 2015-06-29:
uasyncio was announced on both python-tulip and python-ann:

* http://blog.gmane.org/gmane.comp.python.tulip
* https://mail.python.org/pipermail/python-announce-list/2015-June/010799.html

So, let it be as it is.

closed 2015-06-29 by pfalcon

dpgeorge, 2015-06-29:
There are some things that we won't be able to achieve with uPy no matter how hard we try (and yes sometimes due to upstream decisions).

Nice work with announcement!  And nice benchmark results.

BTW, the main reason due to "large" and "unexpected" RAM usage is compiling the scripts, and also the qstrs.  I have a patch in queue for reducing qstr footprint by a decent amount.  Reducing parse tree size could also be worked on.  And I also have a neat script to check exact amount of heap needed by a script: it does binary search on the heapsize, running the script and checking for success :)

pfalcon, 2015-07-02:
> There are some things that we won't be able to achieve with uPy no matter how hard we try (and yes sometimes due to upstream decisions).

Sure, just would like to be sure these aspects were consider, known, and discoverable by other parties at later time.

> BTW, the main reason due to "large" and "unexpected" RAM usage is compiling the scripts, and also the qstrs. I have a patch in queue for reducing qstr footprint by a decent amount. Reducing parse [...]

Looking forward to see these things pushed to mainline, it's particularly interesting to see what you did to qstr beyond what we already have (it's switching to 1-byte hash value, like I suggested in the beginning, right? ;-) ).

danicampora, 2015-07-02:
> Looking forward to see these things pushed to mainline

Looking forward to it as well ;-)