micropython/micropython-lib #41: itertools.cycle fails for generators

stinos, 2015-08-25:
The cycle implementation does not make a copy of the elements iterated over and as such only works for proper sequences like lists, but not for all types of iterables:

```
from itertools import cycle

def reverse(l):
    while l:
        yield l.pop()

for x in cycle(reverse([1, 2, 3])):
    print(x)
```

output:

```
3
2
1

```

Proposed fix: just copy the equivalent python implementation?

```
def cycle(iterable):
    saved = []
    for element in iterable:
        yield element
        saved.append(element)
    while saved:
        for element in saved:
              yield element
```

pfalcon, 2015-08-25:
> Proposed fix: just copy the equivalent python implementation?

Do you mean copy from CPython stdlib? Simple copy is never simple copy, there're also licensing issues to account for.

Also, the above implementation clearly bloats memory in case simple sequence is passed, so I'd suggest to add check for that and avoid copying in that case (I know there're bunch of corner cases, but let's resolve them on being hit). Feel free to submit a patch.

stinos, 2015-08-25:
> Do you mean copy from CPython stdlib?

No I just meant copying python implementation as shown from https://docs.python.org/3/library/itertools.html#itertools.cycle, not the actual C implementation.

dpgeorge, 2015-08-25:
That definition almost mandates the given implementation.  And, as it says, it's expected to use lots of memory.

stinos, 2015-08-26:
Fixed with #42

closed 2015-08-26 by stinos