micropython/micropython-lib #55: collections.deque extend method

PeterKenyon, 2015-12-01:
is this missing or deprecated, whats best way to append an iterable?

PeterKenyon, 2015-12-02:
Not sure whether we should try to emulate python methods exactly 
Deque objects support the following methods:
append(x) OK
appendleft(x) OK
clear() # one liner
count(x) # one liner
extend(iterable) # one liner
extendleft(iterable)
pop() OK
popleft() OK

not so easy....
remove(value)
reverse()
rotate(n)
maxlen

pfalcon, 2015-12-02:
MicroPython's zen is "do more with less". So, I'm going to add .extend(), because it's indeed one-liner. But there's no need to add all methods en-masse. It should happen on a case by case basis, and as long as you're writing your own code (vs trying to run existing), you of course have freedom to choose which methods to use and which not. All in all, anything can be added, as long as there's (good) argumentation why it's needed.

PeterKenyon, 2015-12-02:
Thanks Paul, I sort of thought that's where we? are going, I like it, ("Its not the code you write - its the code you don't right" as I keep telling people).
So the idea is NOT to be totally compatible but to have a compatible subset. OK
I can see that CPython does have the odd feature creep method, but I'm new to all this.

pfalcon, 2015-12-02:
> So the idea is NOT to be totally compatible but to have a compatible subset. OK

Yes, because otherwise we can spend many-many years in development, and then will have a fully CPython-compatible software, which will be just 2 times smaller than it. It's not worth spending so much effort for such a mediocre result. We want to be order of magnitude smaller, and that necessitates being a subset. You can read more about these ideas in https://github.com/micropython/micropython/wiki/ContributorGuidelines

In the meantime, .extend() method was added.

closed 2015-12-02 by pfalcon