micropython/micropython-lib #64: os: musl incompatibility?

roger-, 2016-01-08:
There seems to be an incompatibility with musl related to `errno` (tested on OpenWrt). See details [here](https://forum.openwrt.org/viewtopic.php?pid=305741). 

(Reporting this on someone's behalf.)

pfalcon, 2016-01-08:
Issue with errno compatibility with any libc was fixed a month ago or so. Please make sure that openwrt has up to date packages and that's what users use.

pfalcon, 2016-01-10:
I've just tagged micropython-lib v1.0. So, @roger- , please make sure OpenWRT has micropython 1.5.2 and micropython-lib 1.0, and retest with those, the issues should be fixed. I'm closing this, please comment to reopen.

closed 2016-01-10 by pfalcon

roger-, 2016-01-11:
Ah, seems like the user didn't try the latest version.

Presumably the bug is fixed, will update and test.