micropython/micropython-lib #66: uasyncio timing on Pyboard

peterhinch, 2016-02-20:
core.py uses time.time() in its run_forever() loop. This produces unexpected results on the Pyboard because time.time() returns an integer. This causes the following code to behave incorrectly but understandably for a while.

There is a second problem which occurs after a period, in which the '1.3' coroutine gets repeatedly scheduled very rapidly, before it again settles down. The code (with trivial name changes) works as expected on cPython.
```python
from uasyncio.core import get_event_loop, Sleep
loop = get_event_loop()
def cb(t, s):
    print(s)
    loop.call_later(t, cb, t, s)
loop.call_soon(cb, 1, 'one second')
loop.call_soon(cb, 1.3, '1.3')
loop.run_forever()
```
Output:
```
one second
1.3
one second
1.3
one second
1.3
one second
1.3
................. OK, I get this. repeats for a while. But then:
1.3
1.3
1.3
1.3
1.3
1.3
1.3
1.3
1.3
1.3
................. Numerous repetitions omitted. Then:
one second
1.3
one second
1.3
one second
```

pfalcon, 2016-02-20:
I don't have immediate explanation why it behaves like that. How the code above works on unix port? Does it work the same way as on pyboard, if you use int(time.time()) on it? We'll get to resolving subsecond resolution on pyboard, though I don't know in which way, but first would like to understand why it works incorrectly as is.

peterhinch, 2016-02-20:
It works on the unix port.

peterhinch, 2016-12-06:
This behaviour no longer occurs. An instrumented version demonstrates that the two routines are called with the correct relative frequencies.

closed 2016-12-06 by peterhinch