micropython/micropython-lib #67: distutils stub

mbuesch, 2016-02-24:
Is it desired to have a tiny distutils stub in micropython-lib that implements some library functions of it in a tiny way?
One example would be here:
http://bues.ch/gitweb?p=awlsim.git;a=tree;f=libs/tiny_distutils/distutils;hb=HEAD
This implements distutils.spawn.find_executable() in a very simple way, so my application can use it.
Would that be a candidate for merging into micropython-lib?

pfalcon, 2016-02-24:
I'd say yes, but there're still questions to consider, e.g. would be single "distutil" package, or individual submodules, like "distutils.spawn".

mbuesch, 2016-02-25:
Hm ok. os.environ doesn't seem to be available on micropython, so my implementation currently does not work.

pfalcon, 2016-02-25:
Use os.getenv().

mbuesch, 2016-02-28:
OK, this works. But to correctly implement this, we need os.stat to check whether the final path is a directory or not (Just checking for the X bit is not OK, because directories have that bit set, too). But os.stat does not seem to be implemented. Is there an alternative?
(os.path.isdir() also fails, because it calls os.stat)

pfalcon, 2016-02-28:
os.stat() is of course implemented, where are looking to? os.stat() was implemented since the early days using FFI, in micropython-lib, and recently (well, more than 2 months ago) moved to C-based module, to improve its portability.

mbuesch, 2016-02-28:
```
MicroPython v1.6-88-g57b96a7 on 2016-02-24; linux version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> import os
>>> os.stat()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'module' object has no attribute 'stat'
```

pfalcon, 2016-02-28:
D'oh! Thanks for the report, fixed.

mbuesch, 2016-02-28:
Nice. With this available my implementation of find_executable works as expected.

jonnor, 2024-08-25:
`distutils` has been removed in upstream CPython as of CPython 3.12. So this is perhaps less relevant than before. Maybe better suited for a third-party package than in micropython-lib?

mbuesch, 2024-08-25:
I close this, because distutils is deprecated/removed and this is not relevant for me personally anymore.
Thanks for the discussion!

closed 2024-08-25 by mbuesch