micropython/micropython-lib #79: urequests truncated response on ESP8266

deshipu, 2016-06-19:
I'm trying to use urequests to get weather data from the openweathermap.org API. I'm using the following code:

```
import urequests
r = urequests.get("http://api.openweathermap.org/data/2.5/weather?q=Zurich&appid=XXX").json()
```

Unfortunately, that throws a ``ValueError: syntax error in JSON`` error. Examining the text of the response, I see that it is trucated:

```
{"coord":{"lon":8.55,"lat":47.37},"weather":[{"id":500,"main":"Rain","description":"light rain","icon":"10d"}],"base":"cmc stations","main":{"temp":287.38,"pressure":1022,"humidity":89,"temp_min":282.04,"temp_max":292.45
```

compared to what I get normally with a browser or wget:

```
{"coord":{"lon":8.55,"lat":47.37},"weather":[{"id":802,"main":"Clouds","description":"scattered clouds","icon":"03d"}],"base":"stations","main":{"temp":286.43,"pressure":1022,"humidity":86,"temp_min":282.04,"temp_max":288.71},"wind":{"speed":1.86,"deg":271.002},"rain":{"3h":0.0725},"clouds":{"all":48},"dt":1466327930,"sys":{"type":3,"id":9109,"message":0.035,"country":"CH","sunrise":1466306928,"sunset":1466364353},"id":2657896,"name":"Zurich","cod":200}
```

pfalcon, 2016-06-19:
If you could provide full URL to reproduce (e.g., register a test appid), it would help to investigate this issue quicker.

deshipu, 2016-06-19:
Here's an API key you can use for testing: ``be33ef9a0cde9e4e9b321f0d9ccd75f2``.
Since this issue is public, I will delete it in a few days, though.

pfalcon, 2016-06-19:
Fixed by https://github.com/micropython/micropython/pull/2193 for me.

closed 2016-06-19 by pfalcon

deshipu, 2016-06-19:
Works for me great now. Thank you!

pfalcon, 2016-06-19:
Please test any other networking code you may have with this change, as it's more or less deep one.