micropython/micropython-lib #85: umqtt.simple: multiple subscribe() need check_msg()

puuu, 2016-08-01:
Example script:
```python
import time
from umqtt.simple import MQTTClient

def sub_cb(topic, msg):
    print((topic, msg))

def main(server="localhost"):
    c = MQTTClient("umqtt_client", server)
    c.set_callback(sub_cb)
    c.connect()
    c.subscribe(b"foo_topic")
    #time.sleep_ms(100)
    #c.check_msg()
    c.subscribe(b"bar_topic")
    while True:
        c.wait_msg()

    c.disconnect()
```

Running with mosquitto MQTT broker seems to work fine with:
```console
$ mosquitto_pub -t "foo_topic" -m "1"
$ mosquitto_pub -t "bar_topic" -m "1"
$ mosquitto_pub -t "bar_topic" -m "3"
```

However, having a retained message,
```console
$ mosquitto_pub -t "foo_topic" -m "1" -r
```
the start of the script fails with:
```
Traceback (most recent call last):
  File "example_sub_multi.py", line 21, in <module>
  File "example_sub_multi.py", line 14, in main
  File "/home/frederik/Downloads/micropython-lib/umqtt.simple/umqtt/simple.py", line 104, in subscribe
AssertionError:
```

Having a look in the code shows that `subscribe()` is expecting the
answer of the second subscribe (0x90), but since the message of the
first subscribe arrived, the `AssertionError` is thrown. So,
uncommenting `sleep_ms()` and `check_msg()` results in a working
script.

@pfalcon: This is a bug or by intention?

pfalcon, 2016-08-07:
This is definitely a bug, feel free to propose a patch.

puuu, 2016-08-12:
Solved with #89

closed 2016-08-12 by puuu