micropython/micropython-lib #87: In publish: function takes 2 positional arguments but 3 were given

rudydevolder, 2016-08-07:
Subscribing is working but when I just do:

**c.publish(b"topic",b"This is a test")** I get :
TypeError: function takes 2 positional arguments but 3 were given

How can that be?

pfalcon, 2016-08-07:
There's an example in the repository: https://github.com/micropython/micropython-lib/blob/master/umqtt.simple/example_pub.py . Start with it, see if it works for you, then compare what you do differently in your code.

Also, please avoid cross-posting your question.

dmartauz, 2016-08-11:
I am having similar issues on a host with WiPy port (MicroPython v1.7-153-gbd66b09 on 2016-04-25). 
I renamed simple.py to umqtt.py and uploaded to '/flash/lib'. 

Publish does not work:
>>> from umqtt import MQTTClient
>>> c = MQTTClient("umqtt_client", "xxx.cloudapp.net")
>>> c.connect()
0
>>> c.publish(b"foo_topic", b"hello")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/flash/lib/umqtt.py", line 76, in publish
TypeError: argument num/types mismatch

Subscribing works well:
>>> from umqtt import MQTTClient
>>> def sub_cb(topic, msg):
...     print((topic, msg))
...
...
...
>>> c = MQTTClient("umqtt_client", "xxx.cloudapp.net")
>>> c.set_callback(sub_cb)
>>> c.connect()
0
>>> c.subscribe(b"foo_topic")
>>> while True:
...     c.check_msg()
...
...
...
(b'foo_topic', b'test message from MQTTlens')

Both tests were perform immediately after soft reset.

pfalcon, 2016-08-11:
Ah, I see. umqtt package requires MicroPython 1.8.3 (micropython-lib as a whole generally require the latest MicroPython).

dmartauz, 2016-08-11:
Thanks, I will update my MP host and try it again.

closed 2016-08-23 by pfalcon