micropython/micropython-lib #90: umqtt.simple: query regarding qos == 1

peterhinch, 2016-08-08:
Consider a client using robust.py and publishing with qos == 1 where a network issue prevents the server receiving the message. robust.py traps the error and calls super().connect(False). The [spec](http://docs.oasis-open.org/mqtt/mqtt/v3.1.1/os/mqtt-v3.1.1-os.html#_Toc398718048) section 4.4 states:

"When a Client reconnects with CleanSession set to 0, both the Client and Server MUST re-send any unacknowledged PUBLISH Packets (where QoS > 0) and PUBREL Packets using their original Packet Identifiers [MQTT-4.4.0-1]. This is the only circumstance where a Client or Server is REQUIRED to redeliver messages."

Does this occur with PUBLISH packets? If not, with qos == 1, if a publish message is not received by the server owing to a connection failure the qos contract will not be honoured.

I hope I haven't made an idiot of myself again with this code - it's tricky stuff :)

closed 2016-08-08 by peterhinch

pfalcon, 2016-08-08:
I'm not sure why you closed this ticket and especially deleted all content.

Retry happens at the top level, so entire publish sequence will be repeated (which may be an issue for QoS2, but it's cunningly not supported). The only issue then is that PUBLISH will be retried with *new* packet ID, unlike standard says. That can be fixed, but would require few lines => more memory pressure. So, unless given non-speculative evidence that this is a real problem for esp8266 software which affects many people, I'd consider this "known issue".

peterhinch, 2016-08-09:
> Retry happens at the top level

I eventually figured this out and concluded it was a non-issue; there seems to be no way to delete an issue raised in error. Thanks for the response.