micropython/micropython-lib #94: umqtt.simple publish with qos 1 pid rollover

peterhinch, 2016-08-25:
In continuously running data logging applications the pid value can overflow the 16 bit field. In this instance struct.pack_into constrains the result to 16 bits but produces an illegal value of 0 (MQTT spec section 2.3.1).

I don't know if this is also a valid concern but cPython's struct.pack_into throws an exception on overflow: if the MicroPython implementation ever became conformant the code would fail.

pfalcon, 2016-08-25:
Is it speculative or did you really see mosquitto complain? How does it look? It's not hard to add proper wraparound.

peterhinch, 2016-08-25:
I've not tried a PID of 0. Merely a suggestion explicitly to code a transition from 0xffff to 1.