micropython/micropython-lib #95: HTTPS help

mleonard87, 2016-08-27:
Hi all,

Im trying to make an HTTPS request from an ESP8266 (specifically on a nodemcu). Ive added the urequests module from this repo and am trying the following with no luck. Is HTTPS possible from the ESP8266? Do I need to change something?

```python
>>> import urequests                                                                                                                        
>>> urequests.get('https://www.google.com')                                                                                                 
ssl_handshake_status: -261                                                                                                                  
Traceback (most recent call last):                                                                                                          
  File "<stdin>", line 1, in <module>                                                                                                       
  File "urequests.py", line 101, in get                                                                                                     
  File "urequests.py", line 56, in request                                                                                                  
OSError: [Errno 5] EIO                                                                                                                      
>>>  
```

Thanks, Mike

dpgeorge, 2016-08-27:
In the MicroPython main repo, in the examples/network directory, there are examples of using HTTPS, and I'm pretty sure they work on esp8266.  Try that first to help debug the problem.

mleonard87, 2016-08-29:
I've tried the [example - http_client_ssl.py](https://github.com/micropython/micropython/blob/master/examples/network/http_client_ssl.py) - and it works as you suggested. It appears to be something specific about the URL I am requesting. It works fine in the example using google.com, but changing this to my URL: w9rybpfril.execute-api.ap-southeast-2.amazonaws.com cause the following error:

```python
Address infos: [(2, 1, 0, '', ('52.84.207.208', 443))]
Connect address: ('52.84.207.208', 443)
ssl_handshake_status: -40
Traceback (most recent call last):
  File "<stdin>", line 38, in <module>
  File "<stdin>", line 21, in main
OSError: [Errno 5] EIO
```

Any suggestions? I can't see anything unusual about the certificate.

dpgeorge, 2016-08-30:
Error -40 is SSL_ALERT_HANDSHAKE_FAILURE.  It would likely require some deep debugging to find out what is wrong... You can try running the code on the unix version of uPy, using the ussl socket.  If you get the same error there then it would be easier to debug.

mleonard87, 2016-08-30:
I can reproduce the issue in the unix version. I've found [this post](https://stackoverflow.com/questions/30817876/wget-ssl-alert-handshake-failure) on stackoverflow and I'm guessing is the same issue, the micropython SSL client doesn't support [SNI](https://en.wikipedia.org/wiki/Server_Name_Indication)? As such I believe this is the same issue as micropython/micropython#2062. As per the example on stackoverflow www.course.org fails with -40 and stackoverflow.com works successfully.

My C skills are pretty horrendous and I don't know where ssl_handshake_status is being imported from to debug properly.

pfalcon, 2016-08-30:
ESP8266 uses axTLS as an SSL library. It indeed supports only core TLS features, and may not support various advanced things in use by some sites.

mleonard87, 2016-08-30:
More specifically https://github.com/pfalcon/axtls/tree/67d27df4b5d097e146599fc4fb160a2adcbf5632. I've also found this version [igrr/axtls-8266](https://github.com/igrr/axtls-8266) which has added/modified support for SNI in this commit https://github.com/igrr/axtls-8266/commit/fe4518da8de87a751ff74111884c775152287ae5 I think. Would it be possible to port this across?

pfalcon, 2016-08-30:
@mleonard87 : Feel free to drive that effort. 1st step is clarifying licensing terms of igrr/axtls-8266. Next step is contacting upstream and seeing if they work on SNI themselves, or would be interested to take implementation from igrr/axtls-8266. After that, next step to make will be clear.

makedin, 2016-11-13:
A heads up to anyone wondering if anything is going on around this issue:

I have contacted Cameron Rich of axTLS project and opened an issue regarding the license of [igrr/axtls-8266](https://github.com/igrr/axtls-8266) but I have not yet received a response from either. I will keep you updated.

pfalcon, 2016-11-13:
Great, thanks for taking some actual steps towards rectifying the situation. You probably wanted to link to the specific issue: https://github.com/igrr/axtls-8266/issues/28

makedin, 2016-12-15:
The latest axTLS version, [2.1.0](https://sourceforge.net/projects/axtls/files/2.1.0/), released last monday, seems to have added support for SNI. I have spent some time reading through the code and playing with the "axssl" sample program and it seems to me that it really works.

fmondaini, 2017-06-11:
Hello. I'm trying to use `urequests` and it's still not working:

I've also tried to use the [http_client_ssl](https://github.com/micropython/micropython/blob/master/examples/network/http_client_ssl.py) example and it's not working:
``` 
Connect address: ('172.217.6.238', 443)
ssl_handshake_status: -261
Traceback (most recent call last):
  File "<stdin>", line 4, in <module>
OSError: [Errno 5] EIO
>>> 
```

I'm using the latest (stable) micropython release 1.9.0 for the esp8266 nodemcu.

dpgeorge, 2017-06-12:
> Hello. I'm trying to use urequests and it's still not working:
> 
> I've also tried to use the http_client_ssl example and it's not working:

In general you'd need to give more info to help fix the issue, but I think the problem is that the file you request over HTTPS is too large.  TLS requires that the client support up to 16k chunk sizes, but esp8266 is configured to allow about 4k maximum.  So HTTPS requests will usually only work for small requests.  You can try `urequests.get('https://micropython.org/ks/test.html')`, that should work because it's a small file.

pfalcon, 2017-06-12:
https://google.com broke for esp8266 because they grew their
certificates beyond 4k, certificates size is the first blocker on
establishing connection. (I will grow buffer size to accommodate
google.com case).

fmondaini, 2017-06-12:
@dpgeorge @pfalcon thanks for the tips. I will get more information and let you know. And thank you for the work done on micropython. :1st_place_medal:

pfalcon, 2017-06-14:
ussl was updated and majority of issues were fixed: https://github.com/micropython/micropython/issues/3140 . If any issues, please submit separate tickets with descriptive title and all the details.

closed 2017-06-14 by pfalcon