micropython/micropython-lib #98: [uasyncio] aclose() doesn't remove reader/writer from event loop

danni, 2016-08-30:
reader, writer = await asyncio.open_connection(hostname, port)
    # sending and receiving
    await reader.aclose()
    await writer.aclose()

```
Traceback (most recent call last):
  File "client.py", line 28, in <module>
  File "/Users/danni/.micropython/lib/uasyncio/core.py", line 112, in run_until_complete
  File "/Users/danni/.micropython/lib/uasyncio/core.py", line 104, in run_forever
  File "/Users/danni/.micropython/lib/uasyncio/core.py", line 89, in run_forever
  File "/Users/danni/.micropython/lib/uasyncio/__init__.py", line 28, in remove_reader
KeyError: <value>
```

dxxb, 2016-08-30:
@danni, I think the issue's title is misleading. The file descriptor is removed but aclose() will try to remove it a second time and _del self.objmap[fd]_ will raise a KeyError here https://github.com/micropython/micropython-lib/blob/master/uasyncio/uasyncio/__init__.py#L28.
Probably all instances of _del self.objmap[fd]_ should become _self.objmap.pop(fd, None)_

pfalcon, 2017-03-12:
So, this wasn't actionated, because I'm not sure what's the best way to deal with it. The idea of uasyncio is to stay lead and mean, and while there're explicit read and write streams, it's still a public API fact that they refer to the same underlying object, and it doesn't make sense to close it twice. A writer is the default thing to close (because that's the way to signal the other side of the connection that you've done with sending data; closing reader is the way to abruptly abort a connection). However,  the code above - closing reader, then writer - should work with the current uasyncio version, but not vice versa. I'm not sure that requires fixing, because again, the idea is that it shouldn't be closed twice.

Overall, this will likely need to wait until until uselect module will get support to store "callback data" on its side.