micropython/micropython-lib #100: Hey I want to ask where is the code about xively in esp8266?

michaelllh, 2016-09-08:
Hey ,I want to ask whcih file in esp8266's floder describe the xively ,Thank you for your answer!

dpgeorge, 2016-09-12:
There is no specific support for xively in the code base, but you can easily do it with the urequests module, something like this:
```python
import urequests
def put_to_xively(channel, value):
    json = {"version":"1.0.0", "datastreams":[{"id":str(channel), "current_value":str(value)}]}
    resp = urequests.put("your xively API url", json=json)
    resp.close()
```

michaelllh, 2016-09-12:
Thank you !

发自网易邮箱大师
在2016年09月12日 15:45，Damien George 写道:

There is no specific support for xively in the code base, but you can easily do it with the urequests module, something like this:

import urequests
defput_to_xively(channel, value):
    json = {"version":"1.0.0", "datastreams":[{"id":str(channel), "current_value":str(value)}]}
    resp = urequests.put("your xively API url", json=json)
    resp.close()

closed 2016-10-04 by pfalcon