micropython/micropython-lib #104: umqtt.simple doesnt work with mosquitto

dmarkey, 2016-09-21:
Version: 0.15.-2

>>> client = umqtt.MQTTClient("sddsdsd", "dmarkey.com", 1883)
>>> client.connect()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "umqtt.py", line 84, in connect
IndexError: bytes index out of range

dmarkey, 2016-09-21:
latest micropython.. unix port

phieber, 2016-10-01:
Have you tried to connect from the very same subnet to the broker, using another client like an Android App? Are you using the ESP-01 with 8Mb flash?

andrisarkameru, 2016-10-10:
@dmarkey 
By the provided umqtt.simple samples I think you have to use `b"sddsdsd"` and `b"dmarkey.com"` instead of the strings you provided as your test sample.:

> client = umqtt.MQTTClient(b"sddsdsd", b"dmarkey.com", 1883)

The same applies to the strings you post and topics you subscribe to using the umqtt.simple library.

I recently just started to fiddle with the esp8266 myself and this worked for me.

SquallHalle, 2016-10-12:
Same here. ESP8266:
def FUCK():
    c = MQTTClient(b"umqtt_client", b"10.0.0.5", 1883)
    c.connect()
    c.publish(b"foo_topic", b"hello")
    c.disconnect()

pfalcon, 2016-10-13:
umqtt.simple definitely works with Mosquitto, that's how it was developed and how it's being tested. There're (official, working) examples provided at the same place where umqtt.simple lives: https://github.com/micropython/micropython-lib/tree/master/umqtt.simple

Other related information: http://www.chiark.greenend.org.uk/~sgtatham/bugs.html

phieber, 2016-10-15:
I can also confirm that it works with Mosquitto.
Reliably over serveral days is a different story though ;) But that's where umqtt.robust comes into play

BigMan200, 2016-10-30:
I also do not get it working:
'>> c = umqtt.simple.MQTTClient(b"client", b"10.0.0.32", 1883)
'>>> c.connect()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "umqtt/simple.py", line 84, in connect
IndexError: bytes index out of range

mkarliner, 2016-10-30:
I've taken the liberty of trying this with your (open to the Internet) broker. Sorry for the intrusion.
This works fine for me. Have you imported the MQTT package from your main?
If so, it probably won't work, as the ESP doesn't have enough memory to compile MQTT.
However, that being said, it usually manifests itself as a memory allocation error.
I was using a custom build with MQTT simple frozen in the modules directory of the build.

BigMan200, 2016-10-30:
mkarliner:
did you connect with my mqtt-broker??? If yes: cool and I have no problem with that "intrusion".
BUT: what URL did you use? I mean, above I "only" provided 10.0.0.32 ... which is the "internal" LAN-IP ...

mkarliner, 2016-10-30:
Sorry, I was referring to dmarkey.com in the original post.

timsengit, 2022-09-06:
in my case,I write wrong port case this error,please check your emqttx port one more time!!

mkarliner, 2022-09-06:
Wow that must be some kind of record for length of time between comments!

jonnor, 2024-08-25:
Issue seems to not be reproducible. Proposing to close.

closed 2024-08-25 by mattytrentini