micropython/micropython-lib #109: Packages on pypi are lacking of setup.py

ly0, 2016-10-24:
Using pip-micropython to install micropython's packages will resulting in an error.

```
$ pip-micropython install micropython-cgi
Warning: MICROPYPATH is not set, assuming default value
Destination library directory: /Users/latyas/.micropython/lib
/tmp/pip-micropy-venv/lib/python2.7/site-packages/pip/commands/install.py:180: UserWarning: Disabling all use of wheels due to the use of --build-options / --global-options / --install-options.
  cmdoptions.check_install_build_global(options)
Collecting micropython-cgi
  Downloading http://pypi.doubanio.com/packages/14/a1/397dbc567329e20fbb01c9e8c4f7b122e018368696eba53daa5b0fbc8638/micropython-cgi-3.3.3-2.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
    IOError: [Errno 2] No such file or directory: '/private/tmp/pip-build-Qff7fH/micropython-cgi/setup.py'

    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /private/tmp/pip-build-Qff7fH/micropython-cgi/
```

Seems setup.py is missing on pypi, but exists on this repo.

pfalcon, 2016-10-24:
pip-micropython is deprecated long, long ago. Read MicroPython's README to learn how to install MicroPython packages.

pfalcon, 2016-10-24:
Any confusion should now be fixed by https://github.com/micropython/micropython/commit/b78144c64d71e8b850a0fd7051bcaa9ee0cce8dd.

closed 2016-10-24 by pfalcon