micropython/micropython-lib #111: Unhelpful Exception message on MQTT publish.

mkarliner, 2016-10-30:
If I try to do a publish on a yet to be connected client the following happens.

state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 2
cnt

connected with mihome, channel 6
dhcp client start...
ip:192.168.10.185,mask:255.255.255.0,gw:192.168.10.1

>>> from simple import MQTTClient
>>> c = MQTTClient("adcdd", "192.168.10.195") # Broker not running
>>> c.publish("foo", "Hello MQTT")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "simple.py", line 110, in publish
AttributeError: 'NoneType' object has no attribute 'write'
>>>

It would be nice if the exception was more explicit about the cause.

pfalcon, 2016-10-30:
I don't agree the exception message is unhelpful. Here's an example of unhelpful exception message:

~~~
Fatal exception 29(StoreProhibitedCause):
epc1=0x4000e1c3, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000018, depc=0x00000000
~~~

Compare it with the above - the latter tells exactly what issue happened where. In this case, it's caused by trying to publish to server without having connected first.

mkarliner, 2016-10-30:
Maybe unhelpful was the wrong word, as I said, a more explicit exception would help a less experienced programmer to realise his mistake quicker.

closed 2016-10-30 by mkarliner

pfalcon, 2016-10-30:
I can only refer to https://github.com/micropython/micropython-lib/issues/103#issuecomment-257146384 in the meantime. Excessive error checking goes against MicroPython zen, for very simple reason - it takes  more code, and that means it takes more code, and more memory (which can be alleviated by using frozen bytecode, but then novices don't know about it or how to use it.) So, going too helpful will at some point have opposite effect - people won't be able to use it, because it's too big. As an example, I already found that with all the latest features contributed to umqtt.simple (which initially didn't fit into "simple" definition, but which I couldn't reject, as well, they're useful), it no longer runs in 16K of heap. As umqtt.simple is part of the standard library, it must stay simple and be able to work on as many as possible systems. As the comment linked above says, we welcome (and encourage) community to develop alternative mqtt clients, either based on umqtt.simple or completely new. Thanks for understanding.