micropython/micropython-lib #112: Unconnected MQTT publish fail silently

mkarliner, 2016-10-30:
If I connect up an MQTT client and publish (sucessfully), then stop the broker, subsequent publishes fail without error. I'm not sure if this is the desired functionality or not. In fact, its a bit opaque to me how robust achieves  its reconnection functionality.

connected with mihome, channel 6
dhcp client start...
ip:192.168.10.185,mask:255.255.255.0,gw:192.168.10.1

>>> from simple import MQTTClient
>>> c = MQTTClient("adcdd", "192.168.10.195") # Broker not running
>>> c.publish("foo", "Hello MQTT")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "simple.py", line 110, in publish
AttributeError: 'NoneType' object has no attribute 'write'
>>> c.connect()
0
>>> c.publish("foo", "Hello MQTT")
>>> # Stop broker
>>> c.publish("foo", "Hello MQTT")
>>>#No exception....

pfalcon, 2016-10-30:
> In fact, its a bit opaque to me how robust achieves its reconnection functionality.

It doesn't have to be like that - code is at every MicroPython user's fingertips and simple to understand: https://github.com/micropython/micropython-lib/blob/master/umqtt.robust/umqtt/robust.py#L29

I'll look into the main topic of this issue when time permits.

mkarliner, 2016-10-30:
Sorry, me being unclear. Looking at the code, I wonder how robust would reconnect in a situation other than a publish failure?

pfalcon, 2016-10-30:
Subscription requires running event loop (wait_msg() function). It's also protected in a similar fashion to publish(): https://github.com/micropython/micropython-lib/blob/master/umqtt.robust/umqtt/robust.py#L37

mkarliner, 2016-10-30:
Thanks for the explanation, obvious now you say.

jonnor, 2024-08-25:
The question seems to have been resolved. Proposing to close.

closed 2024-08-25 by robert-hh