micropython/micropython-lib #113: uasyncio  out of date?

mkarliner, 2016-11-02:
https://github.com/micropython/micropython-lib/blob/master/uasyncio/uasyncio/__init__.py

starts 

import errno
import uselect as select

uselect no longer exists in micropython-lib. I fixed this in my copy and then ran up against ffilib and ffi issues, ie:
ffi no longer exists, 
on line 11 of select.py libc.func has no attribute NoneType, etc, etc.

Basically, I'm trying to get the asyncio http server running with a view to being able to write
asyncio network code.  Is there a better branch to be working with?

pfalcon, 2016-11-02:
Which platform do you use? uasyncio supported on unix only so far. Here's an integration test I use to test it:

~~~
cd ~/tmp
micropython -m upip install -p app notes-pico micropython-filedb
MICROPYPATH=app micropython -m notes_pico.main
~~~

Then open link shown and test it.

> uselect no longer exists in micropython-lib

uselect, just like any other `u*` module exists in micropython itself.

mkarliner, 2016-11-02:
Ah. I was trying it on the ESP. Is there canonical list of what is supported 
on the ESP (or any other platform) and what isn't?
Basically, I'm trying to get to some evented/async model on the ESP. Is there a rough
timeframe for asyncio/threads/anything on the ESP?
What can I do to help?

pfalcon, 2016-11-02:
> Is there canonical list of what is supported on the ESP (or any other platform) and what isn't?

Nope, but there's a thread on the forum which warns about that - http://forum.micropython.org/viewtopic.php?f=16&t=2544 . Interested parties can add comments there.

> Is there a rough timeframe for asyncio/threads/anything on the ESP?

It would be next thing I'd work on, after current things are led to some milestone. And it's in that state since ~ July. There're steps definitely being taken though, e.g. http://forum.micropython.org/viewtopic.php?f=2&t=2589 is a refactor needed for that.

> What can I do to help?

There're enough changes for the upcoming 1.8.6 release, and testing the current tree would be the best help. In particular, https://github.com/micropython/micropython/commit/a6c9060d818c38a72ba2758dfaeadd722f2294b8 needs detailed testing with real-world user apps.

mkarliner, 2016-11-02:
OK, I'll switch my builds to 2.0....

pfalcon, 2017-03-03:
Hopefully this can be closed now. If there're specific issue still, please submit them separately (unless alredy submitted).

closed 2017-03-03 by pfalcon