micropython/micropython-lib #115: vernemq compatibility

njordr, 2016-11-06:
Hi.

Have you got on your roadmap the compatibility with vernemq? 

I try your code (publisher) and it fails on connect in this assert (line 84):

    assert resp[0] == 0x20 and resp[1] == 0x02

Thanks

pfalcon, 2016-11-06:
Please read http://www.chiark.greenend.org.uk/~sgtatham/bugs.html and try again.

njordr, 2016-11-06:
vernemq 0.14.1
umqtt.simple (lastest from github)
wipy 2 board

On the wipy board the following code

```
from umqtt.simple import MQTTClient

def main(server="192.168.1.150"):
    c = MQTTClient("ran", server)
    c.connect()
    c.publish(b"test/ran", b"hello")
    c.disconnect()

if __name__ == "__main__":
    main()
```

On the other side  simple consumer in paho

```
import paho.mqtt.client as mqtt  # The callback for when the client receives a CONNACK response from the server.

def on_connect(client, userdata, rc):
    print("Connected with result code " + str(rc))
    # Subscribing in on_connect() means that if we lose the connection and
    # reconnect then subscriptions will be renewed.
    # client.subscribe("$SYS/#")
    client.subscribe('test/ran')

# The callback for when a PUBLISH message is received from the server.
def on_message(client, userdata, msg):
    print(msg.topic + " " + str(msg.payload))

client = mqtt.Client()
client.on_connect = on_connect
client.on_message = on_message
# client.username_pw_set('test01', password='test')

client.connect("192.168.1.150", 1883, 60)

# Blocking call that processes network traffic, dispatches callbacks and
# handles reconnecting.
# Other loop*() functions are available that give a threaded interface and a
# manual interface.
client.loop_forever()

```

When I run the publisher, this is the error on wipy:
Traceback (most recent call last):
  File "<stdin>", line 12, in <module>
  File "<stdin>", line 6, in main
  File "/flash/lib/umqtt/simple.py", line 84, in connect
IndexError: bytes index out of range

In the vernemq error log:
2016-11-06 20:46:13.121 [error] <0.583.0> CRASH REPORT Process <0.583.0> with 0 neighbours crashed with reason: no match of right hand value <<0,4,77,81,84,84,4,2,0,0>> in vmq_parser:parse/4 line 71
2016-11-06 20:46:13.121 [error] <0.583.0> Ranch listener {{0,0,0,0},1883} terminated with reason: {badmatch,<<0,4,77,81,84,84,4,2,0,0>>}

Please let me know if you need more details (apologize to not gave them before)

dpgeorge, 2016-11-07:
Did you try your set-up with mosquitto as the broker?  The code above works for me, running the simple publish code with micropython unix, and the paho code with CPython unix.

And I tested the same thing but with vernemq as the broker.  It also works.

Possibly the error is in the WiPy 2.0 firmware.

njordr, 2016-11-07:
With mosquitto it works. I'll try to write to wipy guys

Thanks

closed 2016-11-07 by njordr