micropython/micropython-lib #124: uasyncio.queue: bug testing queue size in put() function

BrendanSimon, 2016-11-25:
The `put()` function tests for queue size **greater than** max queue size, instead of queue size **greater than or equal to** max queue size.

```
diff --git a/uasyncio.queues/uasyncio/queues.py b/uasyncio.queues/uasyncio/queues.py
index 4a8ae5f..04918ae 100644
--- a/uasyncio.queues/uasyncio/queues.py
+++ b/uasyncio.queues/uasyncio/queues.py
@@ -61,7 +61,7 @@ class Queue:
 
             yield from queue.put(item)
         """
-        while self.qsize() > self.maxsize and self.maxsize:
+        while self.qsize() >= self.maxsize and self.maxsize:
             yield from sleep(self._attempt_delay)
         self._put(val)
```

pfalcon, 2016-12-08:
Thanks for the report. uasyncio.queues was contributed (I didn't look into it myself much), but the fix looks right, especially that in put_nowait() it uses >=. Merging.

@nvbn: FYI

pfalcon, 2016-12-08:
Fixed, thanks.

closed 2016-12-08 by pfalcon