micropython/micropython-lib #128: Endianness issue in re module

Snigelson, 2016-12-07:
First off, I'm not sure if this is an issue in libpcre, micropython or micropython-lib.

I just compiled micropython (version 1.8.3) and libpcre for a big endian MIPS32 Linux system using Buildroot. Everything was working fine until I tried to parse some JSON using the json module from micropyhton-lib.

    >>> json.loads('{"a":"a"}')
    Traceback (most recent call last):
      File "<stdin>", line 1, in <module>
      File "json/__init__.py", line 319, in loads
      File "json/decoder.py", line 343, in decode
      File "json/decoder.py", line 361, in raw_decode
      File "json/decoder.py", line 359, in raw_decode
      File "json/scanner.py", line 37, in _scan_once
      File "json/decoder.py", line 170, in JSONObject
      File "json/decoder.py", line 74, in py_scanstring
      File "re.py", line 81, in match
      File "re.py", line 71, in search
    MemoryError: memory allocation failed, allocating 402653196 bytes

This corresponds to the line (in re.py):

    ov = array.array('i', [0, 0, 0] * (cap_count + 1))

`cap_count` is set on line 69:

    pcre_fullinfo(self.obj, None, PCRE_INFO_CAPTURECOUNT, buf)

using a function from libpcre:

    pcre = ffilib.open("libpcre")
    pcre_fullinfo = pcre.func("i", "pcre_fullinfo", "PPip")

Everything worked great after I added a line to change the endianness of `cap_count`.

I don't know much about ffi libraries so any pointers to where to continue my search would be highly appreciated.

pfalcon, 2016-12-08:
I pushed a fix to the master, can you please try and confirm if it fixed your issue?

Snigelson, 2016-12-08:
Setting byte order in `from_bytes` does not work. Neat solution though! 👍 

    MicroPython v1.6 on 2016-12-07; linux version
    Use Ctrl-D to exit, Ctrl-E for paste mode
    >>> int.from_bytes(b'\x00\x00\x00\x01', 'little')
    16777216
    >>> int.from_bytes(b'\x00\x00\x00\x01', 'big')
    16777216

Note, however, that I am using an older version of Micropython at the moment, as this is the only one I have managed to compile for this system so far. Will get back with results from newer Micropython as soon as possible.

Am I under the right impression, that the Python interpreter is always the same endianness regardless of the system endianness?

Seems like this issue could affect other modules where from_bytes and to_bytes are used. I don't use any of those at the moment so not an issue for me :)

Snigelson, 2016-12-08:
Same result using the latest Micropython on a little endian machine:

    MicroPython v1.8.6-155-g5e39300 on 2016-12-08; linux version
    Use Ctrl-D to exit, Ctrl-E for paste mode
    >>> int.from_bytes(b'\x00\x00\x00\x01', 'little')
    16777216
    >>> int.from_bytes(b'\x00\x00\x00\x01', 'big')
    16777216

pfalcon, 2016-12-19:
Sorry for the delay with fixing this, https://github.com/micropython/micropython-lib/commit/7ccd2d87ed4acae5c13072e0be37ad5b6d6ff7b4 shows how to do that in an alternative way (and why lack of big-endian support for to_bytes/from_bytes is considered low priority).

closed 2016-12-19 by pfalcon