micropython/micropython-lib #129: Typo in urequest module

gadgster, 2016-12-08:
Hello,

I think there's a variable name typo in the urequest module:

```
Index: urllib.urequest/urllib/urequest.py
IDEA additional info:
Subsystem: com.intellij.openapi.diff.impl.patch.CharsetEP
<+>UTF-8
===================================================================
--- urllib.urequest/urllib/urequest.py	(revision 94be568a034643c41f15627f6f191bc5c13265ee)
+++ urllib.urequest/urllib/urequest.py	(revision )
@@ -50,9 +50,9 @@
         l = s.readline()
         if not l or l == b"\r\n":
             break
-        #print(line)
+        #print(l)
         if l.startswith(b"Transfer-Encoding:"):
-            if b"chunked" in line:
+            if b"chunked" in l:
                 raise ValueError("Unsupported " + l)
         elif l.startswith(b"Location:"):
             raise NotImplementedError("Redirects not yet supported")

```
Regards
Richard

pfalcon, 2016-12-08:
Fixed, thanks!

closed 2016-12-08 by pfalcon

gadgster, 2016-12-08:
Hello,

There's a similar issue in urequests module. Patch:

```
Index: urequests/urequests.py
IDEA additional info:
Subsystem: com.intellij.openapi.diff.impl.patch.CharsetEP
<+>UTF-8
===================================================================
--- urequests/urequests.py	(revision 94be568a034643c41f15627f6f191bc5c13265ee)
+++ urequests/urequests.py	(revision )
@@ -81,9 +81,9 @@
         l = s.readline()
         if not l or l == b"\r\n":
             break
-        #print(line)
+        #print(l)
         if l.startswith(b"Transfer-Encoding:"):
-            if b"chunked" in line:
+            if b"chunked" in l:
                 raise ValueError("Unsupported " + l)
         elif l.startswith(b"Location:"):
             raise NotImplementedError("Redirects not yet supported")
```

Regards
Richard

pfalcon, 2016-12-10:
> There's a similar issue in urequests module.

Fixed too, thanks.