micropython/micropython-lib #135: Running external programs, receiving output

Snigelson, 2016-12-15:
There currently does not seem to be any way to run an external program and save the output. Or at least I can't find one. Please correct me if there is one :)

`os.system` only runs the program, prints output to stdout. `os.popen` is not implemented, and all other solutions I found uses the `subprocess` module.

My solution for now is to run a script using `os.system` which writes the output of the program to a file which can then be read from my Python program. Feels like there must be a simpler way.

pfalcon, 2016-12-19:
Yep, weird, I thought popen() was implemented, as I lately seemed to have had a need for it. But what I remember is that I found out that Python's os.popen() doesn't allow to specify text or binary mode. So, maybe I got disappointed with that and didn't go to implement it. And on the other hand, subprocess module is somewhat bloated.

But something of that definitely needs to be implemented.

pfalcon, 2016-12-20:
os.popen() implemented in https://github.com/micropython/micropython-lib/commit/d47419bb73993580eea677b548b1ce526845ebd2

closed 2016-12-20 by pfalcon