micropython/micropython-lib #136: Here is an error in umqtt.simple.py line 139

xiaocaiyidie, 2016-12-19:
line 139 self.sock.write(qos.to_bytes(1)) raise error

error details: 

TypeError: function missing 1 required positional arguments

and i find some information about this function:

int.to_bytes(length, byteorder, *, signed=False)
```python
>>> (1024).to_bytes(2, byteorder='big')
b'\x04\x00'
>>> (1024).to_bytes(10, byteorder='big')
b'\x00\x00\x00\x00\x00\x00\x00\x00\x04\x00'
>>> (-1024).to_bytes(10, byteorder='big', signed=True)
b'\xff\xff\xff\xff\xff\xff\xff\xff\xfc\x00'
>>> x = 1000
>>> x.to_bytes((x.bit_length() // 8) + 1, byteorder='little')
b'\xe8\x03'
```

but when run  `(1024).to_bytes(2, byteorder='big')` in micropython on esp8266, it still can not work.
```python
TypeError: function does not take keyword arguments
```
what wrong with it?  Thanks!

xiaocaiyidie, 2016-12-19:
When i comment this line ,  it can work throght. But I can't receive message published by another client whitch running on my computer(ubuntu 16.04)

dpgeorge, 2016-12-19:
There was a recent fix to MicroPython to require the "byteorder" parameter in the `to_bytes` and `from_bytes` functions (to make it have the same behaviour as CPython).  This change was not propagated (yet!) to all the libraries in this repo.

To fix it for now, use the method without the keyword argument, like this: `qos.to_bytes(1, 'little')`

xiaocaiyidie, 2016-12-19:
Thanks! I have tried it, and it works.

pfalcon, 2016-12-19:
Fixed in master/latest release.

closed 2016-12-19 by pfalcon