micropython/micropython-lib #139: uasyncio repeatedly schedules same coro

peterhinch, 2016-12-21:
The following behaves as expected under CPython. Under MicroPython (Unix) coros are scheduled multiple times to the exclusion of others.
```python
try:
    import asyncio
except ImportError:
    import uasyncio as asyncio

flag = False

class Foo():
    def __init__(self, n):
        self.n = n

    def __await__(self):
        print('Await start', self.n)
        while not flag:
            print('Await', self.n)
            yield
        print('Await done', self.n)

    __iter__ = __await__  # workround for issue #2678

async def run(n):
    foo = Foo(n)
    await foo

async def trig():
    global flag
    print('Trig sleeping for 10s')
    await asyncio.sleep(10)
    print('Trig triggering')
    flag = True

async def rr(n):
    while True:
        print('Roundrobin ', n)
        await asyncio.sleep(0)

loop = asyncio.get_event_loop()
loop.create_task(rr(1))
loop.create_task(rr(2))
loop.create_task(run(1))
loop.create_task(run(2))
loop.create_task(trig())
loop.run_forever()
```
Outcome on CPython before the trigger (sequence repeats):
```
Roundrobin  1
Roundrobin  2
Await 1
Await 2
```
CPython  after the trigger (sequence repeats):
```
Roundrobin  1
Roundrobin  2
```
MicroPython before the trigger:
```
Await 1
Await 2
Roundrobin  2
[repeats 35 times]
Await 1
Await 2
Roundrobin  1
[repeats 35 times]
```
MicroPython after the trigger: Roundrobin 1 is scheduled about 40 times in succession before Roundrobin 2 gets repeatedly scheduled.

pfalcon, 2016-12-22:
Please retest against latest upy/uasyncio.

And feel free to produce the most minimal testcase to show the problem.

peterhinch, 2016-12-23:
Not fixed, I'm afraid.

The following test case demonstrates the problem (MicroPython v1.8.1-1036-gea426dd-dirty on 2016-12-23; linux version). asyncio library updated this morning.

killer() saves having to issue ctrl-C. It isn't necessary to reproduce the issue.
```python
try:
    import asyncio
except ImportError:
    import uasyncio as asyncio

async def killer(duration):
    await asyncio.sleep(duration)

async def rr(n):
    while True:
        print('Roundrobin ', n)
        await asyncio.sleep(0)

loop = asyncio.get_event_loop()
loop.create_task(rr(1))
loop.create_task(rr(2))
loop.run_until_complete(killer(1))
```

peterhinch, 2016-12-23:
On the Pyboard the behaviour is closer to that of CPython. Is it timing related? Unix testing was on a fast PC.

pfalcon, 2016-12-23:
The issue looks like related to the fact that (Python's?) heap sort is not stable. So, if 2 tasks are scheduled at the same time, and reschedule themselves quickly, it could that that one of tasks always ends up on the top of the queue. This hypothesis needs to be confirmed.

peterhinch, 2016-12-24:
A comment in modutimeq.c says that the algorithm is based on heapq, and the [documentation](https://docs.python.org/3/library/heapq.html) for the latter says that the sort is not stable (para 8.5.1). The next para suggests ways round this. A hacker might just use microseconds :)

pfalcon, 2016-12-24:
Specific issue with the code above was fixed yesterday. General issue of unfairness of usyncio scheduling is tracked elsewhere.

closed 2016-12-24 by pfalcon

peterhinch, 2016-12-25:
This works fine until you mix roundrobin coros with awaitable objects. The scheduling then becomes nondeterministic. On some runs (on Unix), sharing is good (but not perfect). On others it's well adrift. Here is my test case:
```python
try:
    import asyncio
except ImportError:
    import uasyncio as asyncio

class Foo():
    def __init__(self, n):
        self.n = n

    def __await__(self):
        while True:
            print('Await', self.n)
            yield

    __iter__ = __await__  # workround for issue #2678

async def run(n):
    foo = Foo(n)
    await foo

async def rr(n):
    while True:
        print('Roundrobin ', n)
        await asyncio.sleep(0)

loop = asyncio.get_event_loop()
loop.create_task(rr(1))
loop.create_task(rr(2))
loop.create_task(run(1))
loop.create_task(run(2))
loop.run_forever()
```
The outcome varies but I have observed the following, where Initially Roundrobin 2 never gets scheduled. Then behaviour changes and Await 1 is ignored.:
```
Roundrobin  1
Await 1
Await 2
Roundrobin  1
Await 1
Await 2
Roundrobin  1
Await 1
Await 2
Roundrobin  1
Await 1
Await 2
Roundrobin  1
Await 1
Await 2
Roundrobin  1
Await 1
Await 2
Roundrobin  1
Await 1
Await 2
Roundrobin  2
Roundrobin  1
Await 2
Roundrobin  2
Roundrobin  1
Await 2
Roundrobin  2
Roundrobin  1
Await 2
Roundrobin  2
Roundrobin  1
Await 2
Roundrobin  2
Roundrobin  1
Await 2
Roundrobin  2
Roundrobin  1
Await 2
Roundrobin  2
Roundrobin  1
Await 2
Roundrobin  2
Roundrobin  1
Await 2
Roundrobin  2
Roundrobin  1
Await 2
Roundrobin  2
Roundrobin  1
Await 2
Roundrobin  2
Roundrobin  1
^C
```