micropython/micropython-lib #144: umqtt on PC

drasko, 2017-01-14:
Hello,
I can see that umqtt is using `localhost` in the examples: https://github.com/micropython/micropython-lib/blob/master/umqtt.simple/example_sub.py#L11

Does this mean that it can be run on the PC?

I am trying to run it and I am getting:
```
micropython pc_mqtt.py 
Traceback (most recent call last):
  File "pc_mqtt.py", line 30, in <module>
  File "pc_mqtt.py", line 14, in main
  File "/home/drasko/.micropython/lib/umqtt/simple.py", line 56, in connect
OSError: 97
```

dpgeorge, 2017-01-16:
> Does this mean that it can be run on the PC?

Yes.

> I am trying to run it and I am getting:

The problem is that your PC supports IPv6 and getaddrinfo is returning the IPv6 address of localhost.  But the socket is created for the IPv4 family.  Connecting that IPv4 socket to an IPv6 address is failing.

The work-around is to use "0.0.0.0" as the address of the server (forcing an IPv4 lookup), instead of "localhost".

drasko, 2017-01-16:
@dpgeorge thanks for the explanation, I confirm that it works.

closed 2017-01-16 by drasko

owendelong, 2018-03-26:
I would argue that the real solution is to update the socket code to work with IPv6 rather than hard-coding the application to IPv4.