micropython/micropython-lib #146: MQTT subscribe() freezes the esp8266 module

akrv, 2017-01-22:
I have a ESP8266 board. 
This scenario is tested both in version 1.8.6 and 1.8.7.

- I have a MQTT broker in the cloud
- ESP connects over wifi and subscribes to broker
- when there is a message in the topic, it toggles a pin in the callback `sub_cb()`

Issue: after running for sometime, the module doesn't respond to the messages. 
So I tried `micropython.mem_info()` and I could see a pattern on of free space decreasing at every loop. 
I do not know if this could be the issue.
It is basically for a use case, when a switch needs to be turned on - the fronted publishes a message and the node reacts. The reaction time is impeccable, but it freezes after sometime.

`c = MQTTClient(CLIENT_ID, server)`
`c.set_callback(sub_cb)  `
`c.connect()  `
`c.subscribe(TOPIC)  `
`while 1:  `
`⋅⋅⋅ c.check_msg()  `
`⋅⋅⋅ time.sleep(1)  `

dpgeorge, 2017-01-22:
Please see this issue: https://github.com/micropython/micropython/issues/2568

jonnor, 2024-08-25:
Duplicate of closed issue. Proposing to close.