micropython/micropython-lib #148: Merging uaiohttpclient

pfalcon, 2017-02-05:
I looked thru my repos and saw https://github.com/pfalcon/micropython-uaiohttpclient which I almost forgot about. So, I propose to merge it into micropython-lib. It's also the easiest way to achieve its installability on esp8266 (gzip-4k packing).

@dpgeorge ^

dpgeorge, 2017-02-06:
I saw there is a asyncio version of requests https://github.com/rdbhost/yieldfromRequests, and it would be nice to support/promote a consistent set of libraries, eg both requests and it's yield-from version.  OTOH, aiohttp looks better maintained than yieldfromRequests, and has more github stars, so perhaps aiohttp is the one we should use.

pfalcon, 2017-11-26:
Imported.

closed 2017-11-26 by pfalcon