micropython/micropython-lib #150: cpython-uasyncio: broken by gzip 4k conversion

ThunderEX, 2017-02-09:
commit dfe4dee62a662793709dbc7c0531580ba5a2371b and 65fb3707ba4d35068971f996aa9664cf9ae7be78 break cpython-uasyncio by introducing such line in setup.py:
```
import optimize_upip
```
and thus, cpython report ImportError, cpython-uasyncio 0.2.1 package on PyPI don't include setup.py, causing this package broken.

pfalcon, 2017-03-12:
Ok, so the idea how to fix this is to introduce new `srctype` in metadata.txt for such packages, for example:

~~~
srctype = cpython-wrapper
~~~

Comments on a specific value to use are welcome.

dpgeorge, 2017-03-14:
> Comments on a specific value to use are welcome.

Or `cpython-adaptor`

pfalcon, 2017-03-14:
Well, "cpython-adaptor" sounds too narrow/specialized IMHO.

ThunderEX, 2017-03-14:
I think cpython-uasyncio is just a normal cpython package, it won't/never work with micropython. So just kick it out of micropython-lib repo into another seperate repo makes sense.

jonnor, 2024-08-25:
cpython-uasyncio is no longer relevant, for several years now.

closed 2024-08-25 by robert-hh