micropython/micropython-lib #153: GET request with urequests on ESP8266 results in OSError: -2

MrLeeh, 2017-02-18:
I am trying to send a simple `GET` request to a host. Everything is fine but after the second or third request I will get an OSError: -2. The code I use is simple:

    from urequests import get

    >>> get('http://mrl33h.de/')
    <Response object at 3fff2250>
    >>> get('http://mrl33h.de/') 
    <Response object at 3fff2840>
    >>> get('http://mrl33h.de/')
    Traceback (most recent call last):
      File "<stdin>", line 1, in <module>
      File "urequests.py", line 101, in get
      File "urequests.py", line 51, in request
    OSError: -2

I can not really put my finger on what's wrong here. I'm using MicroPython v1.8.7-7-gb5a1a20a3 from 09.01.2017.

harandK, 2017-02-18:
I think you have to close the Socket. Get returns a response object,  call close() with it.

MrLeeh, 2017-02-18:
Ah I see. This solves the issue. Thank you for that. It is a bit confusing though because I wouldn' t expect to need to close my response object. How about using a context manager instead?

closed 2017-02-18 by MrLeeh

kaihou-kg, 2020-03-11:
> Ah I see. This solves the issue. Thank you for that. It is a bit confusing though because I wouldn' t expect to need to close my response object. How about using a context manager instead?

I have the issue too. Can I ask, how do you close the socket?