micropython/micropython-lib #154: OSError: [Errno 113] EHOSTUNREACH only when executing connect() in main.py

TheMeaningfulEngineer, 2017-02-20:
Hi

For the following code, 

```
import machine
import network
from umqtt.simple import MQTTClient

led = machine.Pin(D1, machine.Pin.OUT)
button = machine.Pin(D0, machine.Pin.IN)

sta_if = network.WLAN(network.STA_IF)
ap_if = network.WLAN(network.AP_IF)

ap_if.active(False)
sta_if.active(True)
sta_if.connect('my-Lan', 'My-passwoed')

client = MQTTClient("esp-button", "192.168.1.199")
client.connect()

while True:
    if button.value() == 1:
        led.value(1)

    else:
        led.value(0)
```

I'm experiencing the error below:

```
Traceback (most recent call last):
  File "main.py", line 17, in <module>
  File "umqtt/simple.py", line 56, in connect
OSError: [Errno 113] EHOSTUNREACH

```
Line 17 is `client.connect()`.

The peculiarity is that the exact same code can be run without errors when imputed step by step into the interactive shell. Actually, as soon as the exception occurs and the control returns to the shell I can run `client.connect()` without errors (returns 0) which is shown in the MQTT broker log.

The version used is:  `MicroPython v1.8.7-7-gb5a1a20a3 on 2017-01-09` on ESP8266

TheMeaningfulEngineer, 2017-02-20:
This seems to be a No route to host error. [Ref.](https://github.com/micropython/micropython/blob/master/py/mperrno.h#L79) and the mentioned [line 56](https://github.com/micropython/micropython-lib/blob/master/umqtt.simple/umqtt/simple.py#L56) is `self.sock.connect(self.addr)`.

Am assuming this doesn't clasify as a umqtt issue

closed 2017-02-20 by TheMeaningfulEngineer

marcwagner, 2017-03-20:
as a helpfull asside: i ran into this problem as well. turns out the chip needs a few seconds to connect to the wifi. put a time.sleep(3) after the code to connect to your wifi network and you're good to go.

rm534, 2017-10-20:
I run into this too now and cannot for the life of me solve the client.connect() issue. I still keep getting **OSError: [Errno 113] EHOSTUNREACH**.

Any comments on how you solved the problem?