micropython/micropython-lib #155: umqtt simple, __init__ needs wifi to work.

timpur, 2017-02-21:
Ive noticed that to create a new MQTTClient Object you need wifi since in the __init__  this line "self.addr = socket.getaddrinfo(server, port)[0][-1]" needs wifi to resolve the IP address. if there is no wifi the initialization of the new object fails and raises an error.

This can be really annoying when your working with a unstable wifi connection and your trying to code this situation. Wouldn't it be better is this address resolving occurs in the connect function? Every time you connect you also resolve the IP if a host name is used, if an IP was given then there is no need to resolve the IP.

Also little side point when using an host name like 'mqtt.example.com' this get resolved to an IP and stored in memory on the creation of the MQTT object which is the line above. But if the IP of this host name is dynamic then when the IP of the host name changes the MQTT object will never get the new IP and will never be able to reconnect unitl the object is recreated. This make long term connection for people using dynamic DNS for their WAN solution problematic. This can also be fixed by the above fix. moving this line to the connect function.

I think its silly that to create an MQTT object you need wifi to do so. 
Any thoughts on this ?

dmascord, 2017-03-30:
Hi Timpur,

I agree with you.  I have tested the following patch, and it works for me:

```
diff --git a/umqtt.simple/umqtt/simple.py b/umqtt.simple/umqtt/simple.py
index e3b5835..ab5572f 100644
--- a/umqtt.simple/umqtt/simple.py
+++ b/umqtt.simple/umqtt/simple.py
@@ -13,7 +13,8 @@ class MQTTClient:
             port = 8883 if ssl else 1883
         self.client_id = client_id
         self.sock = None
-        self.addr = socket.getaddrinfo(server, port)[0][-1]
+        self.server = server
+        self.port = port
         self.ssl = ssl
         self.ssl_params = ssl_params
         self.pid = 0
@@ -53,6 +54,7 @@ class MQTTClient:

     def connect(self, clean_session=True):
         self.sock = socket.socket()
+        self.addr = socket.getaddrinfo(self.server, self.port)[0][-1]
         self.sock.connect(self.addr)
         if self.ssl:
             import ussl

```

Cheers,

Damien

pfalcon, 2017-03-30:
@dmascord : That would be a duplicate of https://github.com/micropython/micropython-lib/pull/161. Please cooperate with @jcea who submits a proper patch, and let's have it merged.

timpur, 2017-03-31:
I left a comment over there. 
Either solution solves my problem.

puuu, 2017-06-28:
This can be closed? IMHO, solved in #191 .

closed 2017-06-28 by pfalcon