micropython/micropython-lib #156: http.client on esp8266 libpcre related issue?

mhanuel26, 2017-02-24:
Dear All,

While trying to use the http.client library for esp8266, I got the following error while importing email.parser module,

`>>> import email.parser                                                                                                                               
Traceback (most recent call last):                                                                                                                    
  File "<stdin>", line 1, in <module>                                                                                                                 
  File "parser.py", line 12, in <module>                                                                                                              
  File "feedparser.py", line 24, in <module>                                                                                                          
  File "re.py", line 11, in <module>                                                                                                                  
AttributeError: 'NoneType' object has no attribute 'func'`

Then I notice the error is reproduced when I import the re module. 

`>>> import re                                                                                                                                         
>>> help(re)                                                                                                                                          
object <module 're'> is of type module                                                                                                                
  sys -- <module 'sys'>                                                                                                                               
  __name__ -- re                                                                                                                                      
  pcre -- None                                                                                                                                        
  array -- <module 'array'>                                                                                                                           
  ffilib -- <module 'ffilib'> `

pcre is NoneType. Not quit sure about the issue but I know the following is not working 

pcre = ffilib.open("libpcre")

Someone please can let me know if this is a bug or  what am I missing.

Thank you in advance,

dpgeorge, 2017-02-24:
These libraries are not intended to be run on esp8266, they were designed for the unix port of MicroPython.

I'd suggest you try using urequests for a simple http client library (it's in this repo and is known to work on esp8266, in fact it's included in release builds of the port).

mhanuel26, 2017-02-24:
Hi Damien,

Thank you for your quick response,

Cheers,

closed 2017-02-24 by mhanuel26