micropython/micropython-lib #157: check_msg() fails when subscribed message has QOS

diginfo, 2017-02-28:
If a message was sent with the QOS set and the client subscribes to the topic, the client subsequently fails when calling check_msg().

TheSolutionist, 2017-04-05:
Think i'm having the same issue.

Traceback (most recent call last):
  File "main.py", line 60, in <module>
  File "umqtt/simple.py", line 84, in connect
IndexError: bytes index out of range
MicroPython v1.8.7-7-gb5a1a20a3 on 2017-01-09; ESP module with ESP8266
Type "help()" for more information.

diginfo, 2017-04-05:
@pfalcon - I am not sure what additional info you need for this ?

1. Subscribe to Topic on Broker
2. Using another Client, publish a message with QOS = 2
3. mqtt-simple crashes while checking message:

        checking...
        Traceback (most recent call last):
          File "<stdin>", line 1, in <module>
          File "dev/mqtest.py", line 32, in <module>
          File "dev/mqtest.py", line 20, in connect
          File "/lib/umqtt_simple.py", line 192, in check_msg
          File "/lib/umqtt_simple.py", line 185, in wait_msg
        AssertionError:

dpgeorge, 2017-04-07:
Running the unix version of uPy, using the provided example_sub.py code in this repo (both blocking and non-blocking mode), and publishing a message with QoS 0, 1 or 2 works fine for me (the message is received).

@diginfo you'll need tos pecify your platform, the code you are using, and the version of umqtt (because line 185 is blank in the current version so can't have an AssertionError).

diginfo, 2017-04-07:
Thanks, I reported this about 6 weeks ago and have not updated to the latest code, maybe it has been fixed - later today I will test the latest version but can't do that right now.

It was (and still is) working flawlessly when the QOS was not set, but as soon as I set the QOS and sent a message from another client the 8266 crashed.

I just checked and the line in question is line 184, I probably added a comment into my copy of this file which shifted everything down a line.

This is my class code snippet that connects and checks for messages:

		self.client.set_callback(self.subcb)
		
		def subcb(self,topic,msg):
			... do stuff

		def connect(self):
		  for i in range(3):
		    try:
		      self.client.connect()
		      if(self.debug): print(gv.config['broker_host'] + ' connected.')
		      self.client.subscribe(gv.config['get_topic'])
		      time.sleep_ms(100) ## IMPORTANT
		      self.client.check_msg()
		      break
		    except Exception as err:
		      self.error(err)
		      time.sleep(2)

diginfo, 2017-04-07:
Later I will put together a complete example rather than just bits & pieces

diginfo, 2017-04-14:
I am closing this issue as having created some test code, I am now unable to replicate the issue with QOS 1 or 2.

Sorry about that.

closed 2017-04-14 by diginfo