micropython/micropython-lib #162: [Errno 103] ECONNABORTED MQTT

yigibihe, 2017-03-20:
Hello,

I flashed the last stable of micropython to my ESP-01 1mb with esptool. I connected to it the wifi, I can ping the esp from my mqtt server. 

Esp is fed from external power and extra 1000uf capacitor was added for extra stability. I reflashed the module.

```
>>> print(sta_if.ifconfig())
('192.168.4.55', '255.255.255.0', '192.168.4.1', '192.168.4.1')
>>> from umqtt.simple import MQTTClient
>>> c = MQTTClient("umqtt_client", "192.168.4.10")
>>> c.connect()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "umqtt/simple.py", line 56, in connect
OSError: [Errno 103] ECONNABORTED
>>> c.ping()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "umqtt/simple.py", line 94, in ping
OSError: [Errno 103] ECONNABORTED
>>> import esp
>>> esp.check_fw()
size: 586684
md5: 2c7ce68f0e1a5690257698e5e42caf16
True
```

dmascord, 2017-03-30:
What MQTT broker is running on 192.168.4.10 ?   umqtt.simple implements MQTT Version 3.11.  Have you tested the broker with another MQTT client with version 3.11 enabled ?

marev3, 2017-08-31:
Hello,

same here.

on my esp8266 with micropython 1.9.2 I tried:
```
from umqtt.simple import MQTTClient
c = MQTTClient("esp8266", "192.168.1.116")
c.connect()
```

and got:
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "umqtt/simple.py", line 56, in connect
OSError: [Errno 103] ECONNABORTED
```
The server running on 192.168.1.116 is mosquitto 1.3.4 (which should support mqtt 3.1.1) and is working with another client (publishing and subscribing).

What else can I try?

dmascord, 2017-08-31:
Could you try enabling authentication on mosquitto ?  It could be that there is a malformed MQTT packet when authentication is not on

marev3, 2017-08-31:
In the meantime I installed the newest version 1.4.14 of mosquitto that didn't change that error.

> Could you try enabling authentication on mosquitto ?

How do I configure that?

dmascord, 2017-08-31:
https://medium.com/@eranda/setting-up-authentication-on-mosquitto-mqtt-broker-de5df2e29afc
seems like a decent howto

marev3, 2017-09-01:
Okay, I enabled authentication on the mosquitto server, that is working fine.

On the esp8266 I tried it with:
```
from umqtt.simple import MQTTClient
c = MQTTClient(client_id = "esp8266", server = "192.168.1.116", port = 1883, user = b"hallo", password = b"test")
c.connect()
```

and still get:
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "umqtt/simple.py", line 56, in connect
OSError: [Errno 103] ECONNABORTED
```

marev3, 2017-09-02:
I found out that I accidentally had connected the esp8266 to my guest wifi instead of the wifi in which my mosquitto server is.
So my problem is solved.

TJaenichen, 2023-03-21:
Thanks... that could have taken a while

jonnor, 2024-08-25:
Problem solved, issue should be closed.

closed 2024-08-25 by mattytrentini