micropython/micropython-lib #165: incompatible remaining_length in connect

willee, 2017-04-13:
Hi,
    In  CONNECT Packet fixed header, The Remaining Length is encoded using a **variable** length encoding scheme which uses a single byte for values up to 127. Larger values are handled as follows. The least significant seven bits of each byte encode the data, and the most significant bit is used to indicate that there are following bytes in the representation. Thus each byte encodes 128 values and a "continuation bit". The maximum number of bytes in the Remaining Length field is four.

Size of Remaining Length
#Byte x From=>To
#1:0(0x00)=>127(0x7F);
#2:128(0x80,0x01)=>16383(0xFF,0x7F
#3:16384(0x80,0x80,0x01)=>2097151(0xFF,0xFF,0x7F)
#4:2097152(0x80,0x80,0x80,0x01)=>268435455(0xFF,0xFF,0xFF,0x7F)

    In connect, Remaining Length feild is only msg[1],only one  bytes, which is incompatible Remaining Length larger than 127 bytes

Willee Gong

willee, 2017-04-13:
Hi,
    One workaroud:

    def connect(self, clean_session=True):
        print("MQTTClient connect start")
        self.sock = socket.socket()
        self.sock.connect(self.addr)
        if self.ssl:
            import ussl
            self.sock = ussl.wrap_socket(self.sock, **self.ssl_params)
        msg = bytearray(b"\x10") 
        remaining_length = 10 + 2 + len(self.client_id)
        ConnectFlags = clean_session << 1
        if self.user is not None:
            remaining_length += 2 + len(self.user) + 2 + len(self.pswd)
            ConnectFlags |= 0xC0
        if self.lw_topic:
            remaining_length += 2 + len(self.lw_topic) + 2 + len(self.lw_msg)
            ConnectFlags |= 0x4 | (self.lw_qos & 0x1) << 3 | (self.lw_qos & 0x2) << 3
            ConnectFlags |= self.lw_retain << 5
        offset = 1
        #1:0(0x00)=>127(0x7F);
        #2:128(0x80,0x01)=>16383(0xFF,0x7F
        #3:16384(0x80,0x80,0x01)=>2097151(0xFF,0xFF,0x7F)
        #4:2097152(0x80,0x80,0x80,0x01)=>268435455(0xFF,0xFF,0xFF,0x7F)
        assert remaining_length < 268435455
        while True:
            byte = remaining_length % 128
            remaining_length = remaining_length // 128
            if remaining_length > 0:
                byte = byte | 0x80
            msg.extend(struct.pack("!B", byte))
            print('MQTTClient connect msg[',offset,']:',hex(byte))
            if remaining_length == 0:
                break
            offset=offset+1
        msg.extend(b"\0\x04MQTT\x04")
        msg.extend(struct.pack("!B", ConnectFlags))
        if self.keepalive:
            assert self.keepalive < 65536
            msg.extend(struct.pack("!B", self.keepalive >> 8))
            msg.extend(struct.pack("!B", self.keepalive & 0x00FF))

        self.sock.write(msg)

Willee Gong

pfalcon, 2017-04-13:
Duplicate of https://github.com/micropython/micropython-lib/pull/163

closed 2017-04-13 by pfalcon