micropython/micropython-lib #166: [uasyncio] sleep() gets woken up early

klapauciusisgreat, 2017-04-18:
I can reproduce a case where a server wakes up a coro that wants to sleep early, it seems every time the serve handler yields. It only seems to wait up a single coro. I *think* this is a bug, but maybe my async-fu is not what it used to be. Example below (mostly copy/paste from tutorial):

=========================================
import uasyncio as asyncio
import logging

async def serve(reader, writer):
    print(reader, writer)
    print("================")
    print((yield from reader.read()))
    yield from writer.awrite("HTTP/1.0 200 OK\r\n\r\nHello.\r\n")
    print("After response write")
    yield from writer.aclose()
    print("Finished processing request")

async def foo(delay_secs):
    i = 0
    while True:
        print('Hello, ' + str(i))
        yield from asyncio.sleep_ms(delay_secs * 1000)
        i+=1
logging.basicConfig(level=logging.DEBUG)

loop = asyncio.get_event_loop()

loop.create_task(asyncio.start_server(serve, "192.168.2.67", 8888))
loop.create_task(foo(10))

loop.run_forever()
loop.close()
=========================================

As shown below, foo gets called a number of times within the sub-second it takes to handle the web request:

WebREPL connected                                                                               
>>> import main_uasyncio                                                                        
DEBUG:asyncio:start_server(192.168.2.67, 8888)                                                  
DEBUG:asyncio:start_server: Before accept                                                       
Hello, 0                                                                                        
Hello, 1                                                                                        
DEBUG:asyncio:start_server: After iowait                                                        
<StreamReader <socket state=2 timeout=0 incoming=0 off=0>> <StreamWriter <socket state=2 timeout
=0 incoming=0 off=0>>                                                                           
================                                                                                
DEBUG:asyncio:start_server: Before accept                                                       
Hello, 2                                                                                        
b'GET / HTTP/1.1\r\nHost: 192.168.2.67:8888\r\nUser-Agent: curl/7.51.0\r\nAccept: */*\r\n\r\n'  
After response write                                                                            
Finished processing request                                                                     
Hello, 3                                                                                        
DEBUG:asyncio:start_server: After iowait                                                        
<StreamReader <socket state=2 timeout=0 incoming=0 off=0>> <StreamWriter <socket state=2 timeout
=0 incoming=0 off=0>>                                                                           
================                                                                                
DEBUG:asyncio:start_server: Before accept                                                       
Hello, 4                                                                                        
b'GET / HTTP/1.1\r\nHost: 192.168.2.67:8888\r\nUser-Agent: curl/7.51.0\r\nAccept: */*\r\n\r\n'  
After response write                                                                            
Finished processing request                                                                     
Hello, 5                                                                                        
DEBUG:asyncio:start_server: After iowait                                                        
<StreamReader <socket state=2 timeout=0 incoming=0 off=0>> <StreamWriter <socket state=2 timeout
=0 incoming=0 off=0>>                                                                           
================                                                                                
DEBUG:asyncio:start_server: Before accept                                                       
Hello, 6                                                                                        
b'GET / HTTP/1.1\r\nHost: 192.168.2.67:8888\r\nUser-Agent: curl/7.51.0\r\nAccept: */*\r\n\r\n'  
After response write                                                                            
Finished processing request                                                                     
Hello, 7                                                                                        
DEBUG:asyncio:start_server: After iowait                                                        
<StreamReader <socket state=2 timeout=0 incoming=0 off=0>> <StreamWriter <socket state=2 timeout
=0 incoming=0 off=0>>                                                                           
================                                                                                
DEBUG:asyncio:start_server: Before accept                                                       
Hello, 8                                                                                        
b'GET / HTTP/1.1\r\nHost: 192.168.2.67:8888\r\nUser-Agent: curl/7.51.0\r\nAccept: */*\r\n\r\n'  
After response write                                                                            
Finished processing request                                                                     
Hello, 9                                                                                        
Hello, 10                                                                                       
Hello, 11                                                                                       
Hello, 12                                                                                       
Hello, 13                                                                                       
Hello, 14                                                                                       
Hello, 15                                                                                       
Hello, 16                                                                                       
Hello, 17

Note that I experimented a bit and found that adding a number of coros, only one of the coros gets woken up (and the other coros never get affected. I have not properly diagnosed it, but it looks like that the chosen coro is that one that last yielded before the  server connects.

pfalcon, 2017-04-18:
Which version of uasyncio do you use?

klapauciusisgreat, 2017-04-20:
I checked out master a few weeks ago, running on WeMOS D1 mini. I just synced to master, redeployed and the problem went away (BTW, build instructions that _include_ uasyncio, standard drivers like ssd1306 etc aren't as clear as they could be, I'm planning to at least describe my efforts a bit so others can learn.)

If it's useful, I can try to hound down which change caused the bug, but I'm actually happy it works now.

Thanks for being responsive.

closed 2017-04-20 by klapauciusisgreat

pfalcon, 2017-04-20:
Yes, that's the expected behavior - bug with a behavior like you describe was fixed in master, and now finally release to PyPI as uasyncio.core 1.2.

Docs in general need work, and different people treat what's "clear" differently (my grandma would need instructions starting with "turn on computer" ;-) ). By all means, please share your experiences and tips with other users at http://forum.micropython.org/viewforum.php?f=16 , that will also serve as feedback for maintainers.