micropython/micropython-lib #167: can't install with upip.install("micropython-uasyncio")

thetreerat, 2017-04-20:
when I try to install micropython-uasyncio on a Adafuit esp8266 board with micropython 1.8.7 I get the following stack trace:
```python
upip.install('micropython-uasyncio')
Installing to: /lib/
Warning: pypi.python.org SSL certificate is not validated
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "upip.py", line 205, in install
  File "upip.py", line 195, in install
  File "upip.py", line 152, in install_pkg
  File "upip.py", line 144, in get_pkg_metadata
MemoryError:
```
I can install micropython-uasyncio.core

I have flashed it and try with a clean flash and starting with uasyncio and with uasyncio.core already installed. 

I noticed that there is a install.py but I am confused as to how you use it as you need it on the board to install. 

any help would help
thanks

pfalcon, 2017-04-20:
Well, your system doesn't have enough memory to finish install operation. Try to run the command immediately after reboot. If that won't help, you'll need to just copy file manually (via WebREPL or otherwise). You'll find more help at http://forum.micropython.org/viewforum.php?f=16

thetreerat, 2017-04-20:
I was doing it right after clearing the flash and rebooting. I have copied the __init__.py file and been working with it. 

so it a ram issue? not flash? 

not that it matters since I have the file. 

thanks for the input

pfalcon, 2017-05-05:
Yes, it's RAM issue.

And recently, upip was optimized, I have no problem installing uasyncio:

~~~
>>> upip.install('micropython-uasyncio')
Installing to: /flash/lib/
Installing micropython-uasyncio 1.1.1 from https://pypi.python.org/packages/27/33/0d1f599ffdf522bbbbf69e46669536f07dab69659f515b050222f95f6c71/micropython-uasyncio-1.1.1.tar.gz
Installing micropython-uasyncio.core 1.2.1 from https://pypi.python.org/packages/bd/e1/32cf12d1d186967692eb85175d2b23d7de543230bb2b752aaefede59023e/micropython-uasyncio.core-1.2.1.tar.gz
Installing micropython-logging 0.1.3 from https://pypi.python.org/packages/31/60/604096f5ceed147ac985927072ffd6952c0dd89e0d937a8822ad0c832880/micropython-logging-0.1.3.tar.gz
>>> upip.install('micropython-uasyncio')
Installing to: /flash/lib/
Installing micropython-uasyncio 1.1.1 from https://pypi.python.org/packages/27/33/0d1f599ffdf522bbbbf69e46669536f07dab69659f515b050222f95f6c71/micropython-uasyncio-1.1.1.tar.gz
Installing micropython-uasyncio.core 1.2.1 from https://pypi.python.org/packages/bd/e1/32cf12d1d186967692eb85175d2b23d7de543230bb2b752aaefede59023e/micropython-uasyncio.core-1.2.1.tar.gz
Installing micropython-logging 0.1.3 from https://pypi.python.org/packages/31/60/604096f5ceed147ac985927072ffd6952c0dd89e0d937a8822ad0c832880/micropython-logging-0.1.3.tar.gz
>>> 
~~~

(Repeated twice even.)

closed 2017-05-05 by pfalcon