micropython/micropython-lib #168: umqtt.simple using ssl: check_msg() throws NotImplementedError

MrLeeh, 2017-04-21:
The [`check_msg()`](https://github.com/micropython/micropython-lib/blob/master/umqtt.simple/umqtt/simple.py) function uses [`socket.setblocking()`](https://micropython.org/resources/docs/en/latest/wipy/library/usocket.html#usocket.socket.setblocking) to check for pending messages without blocking. While working fine with normal sockets this leads to an `NotImplementedError` when used with a ssl connection.

To create a ssl connection the [`ussl.wrap_socket()`](https://micropython.org/resources/docs/en/latest/wipy/library/ussl.html#ussl.ssl.wrap_socket) function is used which returns a `ussl._SSLSocket` object. For the `_SSLSocket` class there exists only a method stub for `setblocking()` that throws a `NotImplementedError`. So basically the `_SSLSocket.setblocking()` method needs to be implemented in the `ussl` library for this to work.

pfalcon, 2017-04-21:
> So basically the _SSLSocket.setblocking() method needs to be implemented in the ussl library for this to work.

No, of course no. You're staying in you car ahead of big-big pit on the street with a warning sign in front of it. And now you're saying that to pass the street you need to remove this warning sign.

MrLeeh, 2017-04-21:
Ok, I might have missed something here but can't see what. What is the sign warning me from?

pfalcon, 2017-04-21:
That non-blocking I/O support for SSL sockets is not implemented, and it's pretty complex thing ("big-big pit"), or otherwise it would have been implemented long ago.

MrLeeh, 2017-04-21:
I see...so this is a more complex topic... But is there a workaround?
I need to send regular ping requests to the mosquitto server in order to keep the connection alive. I can't do that with a blocking socket. At the moment all I can do is drop the ssl encryption and use a normal socket.

SlowBro904, 2017-07-24:
MrLeeh could you fork a new thread and send pings? I'm very new at this so apologies if that sounds like an ignorant question, but it seems to me that would work.

MrLeeh, 2017-07-24:
@SlowBro904 with full-grown Python this would be possible I guess. But afaik there is no threading implemented in Micropython, yet.

SlowBro904, 2017-07-24:
Threading is usable in Pycom's uPy, and I have successfully flashed that to
a standard WROOM-32. Theirs is open source.

(It's not completely usable in that state, one must also alter the pins in
a csv file and recompile, but I wanted to see it done as a proof of
concept. Also their threading is not complete, with no way to exit a thread
from the outside. I hacked a workaround using a global variable as a
flag and it works.)

SlowBro904, 2017-07-24:
Oh, and I found this. You /might/ have threading in your implementation after all. Go try it.
https://forum.micropython.org/viewtopic.php?t=1864

chtinnes, 2019-10-22:
As a workaround, check_msg might be modified to use polling:
```
    def check_msg(self, timeout):
        poller = uselect.poll()
        poller.register(self.sock, uselect.POLLIN)
        res = poller.poll(timeout)  # time in milliseconds
        if not res:
            return None
        return self.wait_msg()
```

jonnor, 2024-08-25:
`mqtt.simple` should work with SSL for many years now. If there are issues, please re-open with details on how to reproduce.

closed 2024-08-27 by projectgus