micropython/micropython-lib #169: uasyncio core.py: purpose of the Sleep class

peterhinch, 2017-04-21:
Re lines 94 and 95
```python
                        if isinstance(ret, Sleep):
                            delay = int(arg * 1000)
```
If this code runs (by a coro yielding a Sleep instance) an assertion fail follows:
```
AssertionError: Unknown syscall yielded: <Sleep object at 20006c70> (of type <class 'Sleep'>)
```
What is the intention behind the Sleep class? If yielding an instance is invalid the assertion fail is OK - in which case the above lines are presumably redundant.

Or should the Sleep class be an awaitable subclass of SleepMs with a corresponding adjustment to this logic?

pfalcon, 2017-04-21:
uasyncio files may contain redundant code/lines while it's work in progress. When it will be functionality complete, all redundancies will be removed (maybe). Guys named "git blame" and "git log" know the answers about specific lines/code blocks - someone else would need to consult them to answer your questions, but why wait if you can ask them directly?

peterhinch, 2017-12-20:
Closed as this code has now been removed.

closed 2017-12-20 by peterhinch