micropython/micropython-lib #172: urequests' POST method lacks "Content-Type"

green-dalii, 2017-05-01:
When I used `urequests`  on NodeMCU to POST JSON data to a server,it failed and the server sent back the state info:`{"detail": "Unsupported media type \'\' in request."}'`.
When I checked `urequests` code on github,I found that there is no `Content-Type: application/json`

Here is the code quote:

>     if json is not None:
>         assert data is None
>         import ujson
>         data = ujson.dumps(json)
>     if data:
>         s.write(b"Content-Length: %d\r\n" % len(data))
>     s.write(b"\r\n")
>     if data:
>         s.write(data)

pfalcon, 2017-05-01:
It's nice that you check the source, but don't stop that too soon. In this case, you're looking for "headers" in the source code.

closed 2017-05-01 by pfalcon

green-dalii, 2017-05-01:
Oh,Thank you,I'm sorry to bother you.I will have a try again.