micropython/micropython-lib #174: OSError: [Errno 104] ECONNRESET

jonascasarin, 2017-05-04:
Hey, I've a question about connect method in umqtt lib: In the first time connection, even using robust mqtt, my broker need be on-line. If my server is Down at the moment return this error: 

Traceback (most recent call last):
File "umqtt/simple.py", line 56, in connect
OSError: [Errno 104] ECONNRESET

If my broker is on-line at first connection, and after i put down my server, it keeps working well (trying reconnect).

So, can I treat this error for dont stop my application, or it's a problem in the lib?

garybake, 2017-05-04:
Not sure if this is any help - a comment in some example code

```
    # A port on which a socket listened remains inactive during some time.
    # This means that if you run this sample, terminate it, and run again
    # you will likely get an error. To avoid this timeout, set SO_REUSEADDR
    # socket option.
    s.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEADDR, 1)
```

jonascasarin, 2017-05-04:
Do u mean to change this in the lib?!...
Cause there is nothing like that...
And my basic code is junting subscribing in some topics and i tried used setsockopt in client (mqtt object) and doesnt work.

Just to understand my simple aplication:

`def mqtt():
    global client 
    client = MQTTClient(CONFIG['client_id'], CONFIG['broker'], CONFIG['port'], CONFIG['username'], CONFIG['password'])
    client.DEBUG = True
    client.set_callback(sub_cb)
    client.settimeout = settimeout
    if not client.connect(clean_session=False):
        client.subscribe(b"sensor/{}/{}".format(CONFIG['usercode'],CONFIG['client_id']))
        print("Connected to {}".format(CONFIG['broker']))
    while 1:
        client.check_msg()
        time.sleep(1)
`
If could I treat this error in my application would be better than change something in the lib...u dont think so?!... couldn't stop an aplication because this.

I'll seize to ask if when reconect in my broker it's possible subscribe in topic... because I was noticed that when mqtt client is down (offline) and active reconect mode, block all aplication (including whiles and so on).

garybake, 2017-05-05:
Apologies, I hadn't read it properly and rushed to reply. My original comment is no help and should be ignored.

jonnor, 2024-08-25:
Is this still a problem in the latest version of mqtt.simple ?

Josverl, 2025-11-13:
closed due to no response.
Please do feel free to reopen if needed.