micropython/micropython-lib #175: Getting ENOMEM error after several calls to urequests.post()

jdeltoft, 2017-05-05:
I'm running a nodeMCU setup where every 30 seconds I'm calling urequests.post() to post a bit of data to influxdb running on a raspberry pi. It works great for 5-10 minutes and then it starts getting ENOMEM errors and resets eventually. I tried doing a gc.collect() along with my calls to post but didn't seem to help. Is there any other suggestions anyone can think of? Any debugging I should gather? gc.free_mem etc?

pfalcon, 2017-05-05:
Please see if http://docs.micropython.org/en/latest/esp8266/esp8266/general.html#scarcity-of-runtime-resources rings a bell. If it doesn't, please explain way, and we'll try to improve it.

jdeltoft, 2017-05-05:
Sorry for jumping the gun on this.  Got a reply on the forum that I probably made the common mistake and sure enough...  I forgot to call res.close().  <embarrassed look>

closed 2017-05-05 by jdeltoft

whatdtech, 2022-04-06:
problem is with urequests library, if you can switchback to usocket working fine. Here is a good firebase library based on sockets [micropython-firebase-realtime-database](https://github.com/ckoever/micropython-firebase-realtime-database)

Keelando, 2022-09-22:
Was also struggling with this one, but its actually [outlined in the docs](https://mpython.readthedocs.io/en/master/library/mPython/urequests.html)

>It's mandatory to close response objects as soon as you finished
>working with them. On MicroPython platforms without full-fledged
>OS, not doing so may lead to resource leaks and malfunction.
`r.close()`