micropython/micropython-lib #176: uasyncio: Finalizing naming of a function to schedule a coroutine after a millisecond delay

pfalcon, 2017-05-13:
It's convenient to be able to schedule a coroutine to run, but after some delay. Convenient for what? Well, at least for demonstration purposes - you run the same coroutine against one LED, that the same coro against another LED, but with a delay, etc. Is it useful for much beyond that? Dunno.

Normally, this would be done by creating another (perhaps lambda) coro which would delay and await original coro. Memory hog. So, how it's done currently is that there's `call_later_ms_()` method. It comes from the original `call_later()` upstream method, but as clear, takes delay in ms. Underscore at the end means tentative naming. Also, in upstream, `call_later()` applies only to callbacks, but due to the way uasyncio scheduling works, in it it can apply to both callbacks and coros.

So, time to decide whether we're ok with (ab)using that function for this functionality (incompatible with upstream) and drop the trailing underscore, or invent something else. I thought about adding optional arg to upstream's create_task(). But that's even bigger abuse.

@dpgeorge : Thoughts? (Now that I wrote everything down, sticking with `call_later_ms()` seems a natural choice).

@peterhinch: Comments are welcome too.

peterhinch, 2017-05-14:
I agree with ``call_later_ms()`` for the reasons you give. The option of using the "normal" method remains for people wanting to write code portable to CPython (and less concerned about RAM).

pfalcon, 2017-05-14:
> The option of using the "normal" method remains for people wanting to write code portable to CPython (and less concerned about RAM).

Right, maintaining and improving compatibility with upstream asyncio is still a goal for uasyncio.

dpgeorge, 2017-05-14:
A big effort with developing uasyncio has been getting it to work well with millisecond resolution, so having a public method call_later_ms() seems natural (similar to time.sleep() and time.sleep_ms()).  Allowing the callback to also be a coro is a bit more of an "extension" but seems ok if it can be guaranteed to always work even if the underlying implementation of uasyncio changes in the future (or someone creates some derived event loop).

But something else to think about in this context: is it possible to (easily) derive a new event loop that has time measured in microseconds instead of milliseconds?  And then it would add a method call_later_us().  It looks like the run_forever() method has milliseconds hard coded in it.

pfalcon, 2017-05-14:
> It looks like the run_forever() method has milliseconds hard coded in it.

Where? Only wait() has sleep_ms() hardcoded, and it explicitly described as requiring override if anything. Anything else uses the common timebase, EventLoop.time(). Of course, any such timebase should adhere to the standard MicroPythin timing conventions, e.g. being suitable for ticks_diff() - simply because there're no other guaranteed timing conventions in MicroPython.

pfalcon, 2017-05-14:
> But something else to think about in this context: is it possible to (easily) derive a new event loop that has time measured in microseconds instead of milliseconds?

Well, that's why I submitted this ticket, though I didn't intend to dig that deep. How uasyncio.core was modified to support baremetal targets is that it switched to real-world wrap-around timebase of utime module, while still retaining compatibility with upstream, which takes time in Si units, i.e. seconds. There's a single function to override the timebase: EventLoop.time(). But as mentioned above, the timebase should adhere to the invariant of utime.ticks_diff(), which leaves ticks_us() and ticks_cpu() as choices. 

Then, call_later() should be overriden to convert seconds to that timebase. And what we discuss here, call_later_ms_(), actually accepts interval in timebase units. So, what should we call it? call_later_ticks()? Longer and people will get confused.

pfalcon, 2017-05-14:
And realistically, uasyncio won't be used with any other timebase but milliseconds (not be fore we run on gigahertz CPUs with realtime GC).

pfalcon, 2017-05-14:
Of related matter: with the design described above, "absolute Si time", as used by upstream call_at() method, doesn't make sense. If you look at the current source of uasyncio.core, there're 2 functions call_at() and call_at_(), but implementation is the same, and thus incorrect for call_at(), and can't be made correct. Thus, the proposal to remove call_at(), to avoid confusion a-la https://github.com/micropython/micropython-lib/issues/177, while rename call_at_() to the final name. call_at_ms() would be somewhat confusing, call_at_ticks() would be better, but see for call_later_* controversy.

pfalcon, 2017-05-14:
Note also that one way of evolution of uasyncio may be a complete rewrite in C, then methods in it won't be overridable.

pfalcon, 2017-05-14:
> Allowing the callback to also be a coro is a bit more of an "extension" but seems ok if it can be guaranteed to always work even if the underlying implementation of uasyncio changes in the future (or someone creates some derived event loop).

Yes, that's how it was from the very beginning and the most natural way to deal with things. How it's implemented in CPython (IIRC) is that only callback is a native type for the queue, and otherwise Task object adapts a coroutine to have a callback'ish interface for queue runner. That's of course awful bloat, and in uasyncio, both coros and callbacks are native types for the queue, handles by simple `if`.

peterhinch, 2017-05-15:
> Note also that one way of evolution of uasyncio may be a complete rewrite in C, then methods in it won't be overridable.

What would this offer in exchange for that serious loss of flexibility? Performance is excellent - approximately 240μs task switching overhead by my measurements on the Pyboard. I doubt that any reduction would offer measurable gains on realistic embedded MicroPython applications.

pfalcon, 2017-05-15:
Optimization, first of all for memory use. There's no specific plans for that, the idea was always to find a way to optimize Python code to the needed level. But if nothing else will help to run full-stack webapps on esp8266, that's one of possible further actions.

dpgeorge, 2017-05-16:
> It looks like the run_forever() method has milliseconds hard coded in it.
> 
> Where? Only wait() has sleep_ms() hardcoded, and it explicitly described as requiring override if anything. Anything else uses the common timebase, EventLoop.time().

The delay is hardcoded to be 1000 times the argument given to Sleep: https://github.com/micropython/micropython-lib/blob/master/uasyncio.core/uasyncio/core.py#L97-L98

There is a SleepMs class which also sets the units for the delay: https://github.com/micropython/micropython-lib/blob/master/uasyncio.core/uasyncio/core.py#L99-L100

One could not add SleepUs without editing/overriding run_forever().

The actual scheduling is done with call_later_ms_: https://github.com/micropython/micropython-lib/blob/master/uasyncio.core/uasyncio/core.py#L129

dpgeorge, 2017-05-16:
> Then, call_later() should be overriden to convert seconds to that timebase. And what we discuss here, call_later_ms_(), actually accepts interval in timebase units. So, what should we call it? call_later_ticks()? Longer and people will get confused.

I think call_later_ms() should be named just that, and it schedules using milliseconds (regardless of the underlying time unit).  It can then be overridden (and call_later_us() added) in a derived class that wants to use microseconds as the time base.

dpgeorge, 2017-05-16:
> And realistically, uasyncio won't be used with any other timebase but milliseconds (not be fore we run on gigahertz CPUs with realtime GC).

It's not about microseconds per se, but rather sub-milliseconds.  A delay/resolution of 0.1ms is desirable for lots of applications and, aside from using floats as the argument to ms functions, providing "us" functions is the way do to that.  You wouldn't pass 1us to the function, but you would pass 300us (for example).

With MCUs now doing more than 200MHz that gives roughly 1000 MicroPython bytecodes per millisecond, so sub-millisecond resolution is something you want.

As a related aside, I'm already finding that I need mp_hal_delay_ns() (yes, nanoseconds) at the C level in order to implement usable SPI in software (and fast I2C).  Eg a 1MHz SPI bus (which is still pretty slow) needs 500ns delay between clocks.  On 100+MHz CPUs one really needs a ns delay function to do these kind of things.

pfalcon, 2017-05-16:
> The delay is hardcoded to be 1000 times the argument given to Sleep: https://github.com/micropython/micropython-lib/blob/master/uasyncio.core/uasyncio/core.py#L97-L98

That apparently should go to Sleep constructor, so it (class) can be monkey-patched.

> There is a SleepMs class which also sets the units for the delay: https://github.com/micropython/micropython-lib/blob/master/uasyncio.core/uasyncio/core.py#L99-L100

This is SleepTicks by semantics, as discussed above.

> One could not add SleepUs without editing/overriding run_forever().

Yes, there's no need for that, SleepMs is really SleepTicks.

pfalcon, 2017-05-16:
So, let's again contrast 2 choices:

1. Make a (single) timebase of uasyncio to be better configurable, and make related functions end with `_ticks`.
2. Reinforce millisecond timebase, and assume that someone indeed will need to do patching to add yet another timebase.

From previous comments, @dpgeorge leans towards 2. My re-raising this issue shows that I'm leaning towards 1 (because 2 is scope creep). Let's make a final decision.

dpgeorge, 2017-05-16:
There's a 3rd option, related to 2: to use microseconds as the time base.  That should buy enough resolution for a while to not require any patch or subclassing.

So it's really between using an unspecified time base (could be s, ms, us, ns, cpu-ticks, etc), or a specified one (ms or us).  The former seems more attractive from a generality/reusability point of view.  The latter seems more user-friendly in the sense that when you read SleepMs/call_at_ms/call_later_ms in the code you understand immediately what it's doing, whereas SleepTicks/call_at_ticks/call_later_ticks is a bit abstract and you need to read some other part of the code to know the unit of time.

pfalcon, 2017-05-16:
> There's a 3rd option, related to 2: to use microseconds as the time base. That should buy enough resolution for a while to not require any patch or subclassing.

Sorry, that's too big a change which needs to re-validated. Also, we promise not less than 14 bits of ticks period, and recommend to have not less than 20 (i.e. one million decimal). That means that somewhat guaranteed max duration for microsecond ticks is 1s.

> The former seems more attractive from a generality/reusability point of view. The latter seems more user-friendly in the sense that when you read SleepMs/call_at_ms/call_later_ms in the code you [...]

Yes, that pinpoints it well. So, we should decide whether the goal of uasyncio is to be reusable scheduler, or well-defined fixed function scheduler. It kinda written with reusability in mind, but even people who try to reuse it (which are few), just fork and patch, instead of subclassing. And as soon as we move to optimize it in C, it becomes fixed-function.

dpgeorge, 2017-05-16:
> And as soon as we move to optimize it in C, it becomes fixed-function.

If it's going to have parts moved to C (parts that depend on ms/ticks units) then it makes sense to support a general time base, because patching C is much harder (and less maintainable for the user) than patching Python.

pfalcon, 2017-05-16:
If it's moved to C, it's moved to C completely, with no overridable methods left for user. (As we know, one of the biggest inefficiencies of Python is its dynamic method binding, so rewriting in C, makes sense to cover that too.)

peterhinch, 2017-05-16:
Please consider practical usability. Its underlying performance is great: it can switch tasks in 240μs which seems pretty darn quick to me. A faster timebase is fine so long as we keep it in Python.

As I've pointed out, in some applications, limitations of fundamental performance are dwarfed by lack of precision in the sleep functions and long latency of response to events. While the code is in Python these issues can readily be addressed by subclassing or otherwise.

I appreciate the reasons why changes to fix these issues won't make it into the mainline, but I think a solution which precluded a Python fix would be a big step backward. It's flexibility is one of its greatest assets.

pfalcon, 2017-05-16:
Ok, so finalizing on call_later_ms() which leaves path to later introduce sub-millisecond function and optimize to C.

pfalcon, 2017-05-16:
Changes applied in 61392cb641a2ec903f389f4ca4f46468dbecaccc, ad73ee304370adc3845f0d3627662caf4ecf8ecc, 4fa29d867ae79c14f5e93abea8c709b0fce3bac6.

closed 2017-05-16 by pfalcon