micropython/micropython-lib #178: uasyncio: Location of the docs

pfalcon, 2017-05-14:
With API hopefully finalized in https://github.com/micropython/micropython-lib/issues/176 , the question is about the docs. Do we want to make a precedent and add a micropython-lib module documentation to the main MicroPython docs? Because otherwise, there's simply no incentive to document micropython-lib modules, knowing that people will unlikely to find the docs.

@dpgeorge : ^

dpgeorge, 2017-05-16:
The idea is that libs in this repo follow the CPython API so don't need documenting, but, of course, that's too naive and unrealistic :)

Then, yes, it makes sense to have the docs in one place online, ie docs.micropython.org.  But it would be possible to have the .rst files in this repo (under a docs/ dir) and then (sym)link to them from the main repo (like it's done linking frozen modules).  I would prefer that approach.

pfalcon, 2017-11-02:
In the meantime, initial README-style doc was added: https://pypi.python.org/pypi/micropython-uasyncio/1.2.3

jonnor, 2024-08-25:
Documentation for asyncio is now at https://docs.micropython.org/en/latest/library/asyncio.html

closed 2024-08-27 by projectgus