micropython/micropython-lib #179: uasyncio call_soon() and call_later() fail when arguments are passed

peterhinch, 2017-05-19:
These methods pack multiple args into a tuple, unpacking them before passing them to call_at_(). The latter expects a tuple and hence throws an exception:
```python
import uasyncio as asyncio

def cb(x, y):
    print('x = {} y = {}'.format(x, y))

async def foo():
    loop.call_soon(cb, 5, 6)
    await asyncio.sleep(1)

loop = asyncio.get_event_loop()
loop.run_until_complete(foo())
```
```
Traceback (most recent call last):
  File "<stdin>", line 11, in <module>
  File "uasyncio/core.py", line 129, in run_until_complete
  File "uasyncio/core.py", line 121, in run_forever
  File "uasyncio/core.py", line 85, in run_forever
  File "uasyncio/core.py", line 126, in _run_and_stop
  File "<stdin>", line 7, in foo
  File "uasyncio/core.py", line 32, in call_soon
TypeError: function takes 4 positional arguments but 5 were given
>>>  
```

pfalcon, 2017-05-22:
Damn, now I start to remember what's real difference between call_at() and call_at_()...

pfalcon, 2017-05-22:
So, there was another issue - call_later_ms() expected a tuple of args. It's being made consistent with call_later(), though that's less efficient, so may need to introduce funcs taking tuples...

pfalcon, 2017-05-22:
Fixed.

peterhinch, 2017-05-22:
Previously the run_forever method's last line was
```python
self.call_later_ms_(delay, cb, args)
```
As I understand it, args here is a tuple so this was consistent with the method:
```python
    def call_later_ms_(self, delay, callback, args=()):
        self.call_at_(time.ticks_add(self.time(), delay), callback, args)
```

Now the last line is:
```python
self.call_later_ms(delay, cb, args)
```
with the method being
```python
    def call_later_ms(self, delay, callback, *args):
        self.call_at_(time.ticks_add(self.time(), delay), callback, args)
```
Is this correct?

pfalcon, 2017-05-22:
No, thanks for spotting.

pfalcon, 2017-05-22:
e13e139f51572b3a7de910bf5348a80a56ad35d3 tells the story on that.

closed 2017-05-22 by pfalcon